RobotLocomotion/drake #4354: Drake submodules when using git-worktree: error-prone admin area configuration

tarquasso, 2016-12-03:
The .git-admin directories of all drake submodules are within the submodules, but not at the main repository level. According to SmartGit, this is an "error-prone admin area configuration".

To be specific, when I add an additional worktree to my drake-distro, my git IDE SmartGit (http://www.syntevo.com/smartgit/) displays the following error in regards to 23 of the drake submodules:

```
1/23: The submodule at 'externals/bot_core_lcmtypes' has an error-prone admin area configuration.
For the submodule at externals/bot_core_lcmtypes the .git-admin directory is not located below /home/rkk/drake-distro/.git/worktrees/drake-distro-paramest, but at /home/rkk/drake-distro/.git/worktrees/drake-distro-paramest/modules/externals/bot_core_lcmtypes instead.
This indicates an error-prone configuration and will most likely result in confusion and strange behavior. You should re-clone the submodule from scratch.

2/23: The submodule at 'externals/bullet' has an error-prone admin area configuration.
For the submodule at externals/bullet the .git-admin directory is not located below /home/rkk/drake-distro/.git/worktrees/drake-distro-paramest, but at /home/rkk/drake-distro/.git/worktrees/drake-distro-paramest/modules/externals/bullet instead.
This indicates an error-prone configuration and will most likely result in confusion and strange behavior. You should re-clone the submodule from scratch.

etc... for the remaining 21 repositories
```

It think the submodule git admin areas should be at
`~/drake-distro/.git/worktrees/drake-distro-paramest`
here for my worktree `drake-distro-paramest`
but not at 
`~/drake-distro/.git/worktrees/drake-distro-paramest/modules/externals/<submodule-name>` 

I use git worktree to have several separate instances of my drake-distro, since I have to work on branches with histories far apart. 

Is this a know issue or is it nothing of concern? Have other used git-worktree with drake before?

liangfok, 2016-12-04:
This is the first time I've heard of this issue. I'm wondering if @mwoehlke-kitware has an opinion since I believe he set up Drake's git submodules (apologies if I'm wrong).

david-german-tri, 2017-04-11:
We haven't had anyone report an actual problem related to this, so I'll close as wontfix.  After #3129, the submodules will be removed anyhow.

closed 2017-04-11 by david-german-tri