RobotLocomotion/drake #4355: schunk_gripper.sdf is Referencesing gazebo.material

liangfok, 2016-12-03:
# Problem Definition

The model for the Schunk gripper (`drake-distro-2/drake/examples/schunk_gripper/models/schunk_gripper.sdf`) is referencing `file://media/materials/scripts/gazebo.material`. There are two problems with this:

1. This file does not exist on machines that do not have the Gazebo simulator installed. This is possible since Gazebo is no a dependency of Drake.
2. The current SDF parser does not support the `<script>` and `<uri>` tags within SDF's `<material>` tag.

# Possible Solutions

## Idea 1: Fork the Model and Customize It To Be Compatible with Drake
One idea is to modify `schunk_gripper.sdf` to not reference this material script. This is sub-optimal since that will result in Drake essentially forking this model.

## Idea 2: Add Support for Material Scripts and all Aforementioned Tags and Files
Drake will need to maintain its own copy of Gazebo's materials file or require that Gazebo be installed. Adding support to the existing parser should only be done if we decide not to pursue #4337.

RussTedrake, 2016-12-03:
Can you explain why this is actually a problem.  What's wrong with us simply ignoring the tag (which, as far as I know, is the correct behavior for unsupported tags given the xml standard)

liangfok, 2016-12-03:
The visualization will be wrong, which may become a problem when we start to simulate cameras that need to see the gripper and the objects being gripped.

liangfok, 2016-12-05:
I'm now beginning to recall that @ggould-tri may have created this model by hand using Gazebo's model editor, which may explain why Gazebo's material scripts are being referenced. In this case, it would be nice if Drake's SDF parser could be made compatible with the SDFs outputted by Gazebo's model editor.

ggould-tri, 2016-12-05:
Your recollection is correct.

The Gazebo model editor is the only gui SDF editor that I am aware of, so I think supporting it is pretty much mandatory for any wide adoption of Drake as a general-purpose tool.  That said, it's painful to use, unpredictable in effect, and as you note highly idiomatic in its output.  Probably getting a reliable and pleasant gui for SDF should be on Drake's product roadmap, either by forking Gazebo's or by adding editing capabilities to Director.  Either might make a good R&D or intern project.

The solution for the Schunk SDF in particular is to strip out the offending material, whatever it may be.

A second-order solution would be to have a strict parser mode for Drake that errors on unrecognized or unparsed material, or better yet an sdf serializer that output SDF from Drake's IR of an object (RBT and friends).  This would let us strip out noncoding SDF matter efficiently.  Again, possibly a good intern project, albeit for a very detail-oriented and meticulous intern -- "Serialization Formats Are Not Toys" as Tom Eastman put it.

liangfok, 2017-04-12:
I'm going to close this because we're most likely going to switch to the URDF model based on this comment: https://github.com/RobotLocomotion/drake/issues/5267#issuecomment-291853312

closed 2017-04-12 by liangfok