RobotLocomotion/drake #4359: Mysterious sample times

RussTedrake, 2016-12-04:
This is potentially related to #4325 (which was reopened because the sensitivity to floating point errors persists), but i'm opening this issue to address a bigger API question about sample times.

In the unit tests in #4358, I'm registering two systems: a random source which declares a periodic update at 0.0025, an output port with discrete sampling, and a signal logger which declares it's input timing to be "inherited".   If I were to set up the comparable systems in Simulink, I would naively expect the SignalLogger's DoPublish to be called every 0.0025 seconds... but somehow it gets called at times:
```
0
 0.001
 0.002
0.0035
0.0045
 0.006
 0.007
0.0085
0.0095
 0.011
```
Huh!?  I've confirmed that it's using the RungeKutta2Integrator, which should be a fixed-step integrator.  Again, some of the randomness here could be related to the floating point issues, but I think the API is also broken.

I did notice that SamplingSpec in system_port_descriptor.h currently reads
```
/// SamplingSpec describes whether a port has inherited, continuous, or
/// discrete sampling. Since ports in Drake are not actually sampled, this is
/// only potentially useful for detecting unintended connections at Diagram
/// connection time.
// TODO(david-german-tri, sherm1): Consider just getting rid of this.
typedef enum {
  kInherited = 0,
  kContinuous = 1,
  kDiscrete = 2,
} SamplingSpec;
```
That's very misleading!

david-german-tri, 2016-12-05:
> I would naively expect the SignalLogger's DoPublish to be called every 0.0025 seconds...

When I patch your branch onto master, I can't reproduce the numbers you gave above.  I get:

```
[ RUN      ] TestSignalLogger.GaussianWhiteNoise
0.0025
0.005
0.0075
0.01
0.0125
0.015
0.0175
0.02
0.0225
0.025
0.0275
0.03
0.0325
0.035
0.0375
0.04
0.0425
0.045
0.0475
0.05
```
and so on...

david-german-tri, 2016-12-05:
Contrary to earlier post, I _can_ reproduce this one.  Investigating now.

david-german-tri, 2016-12-05:
OK, I see what's going on.

```
0->0.001 Integrator takes max step size, and publishes at the end of the step.
0.001->0.002 Integrator takes max step size, and publishes at the end of the step.
0.002->0.0025 Integrator steps to the discrete update, and does not publish.
0.0025->0.0035 Integrator takes max step size, and publishes at the end of the step.
0.0035->0.0045 Integrator takes max step size, and publishes at the end of the step.
0.0045->0.005 Integrator steps to the discrete update, and does not publish.
... and so on
```

I agree this probably isn't what we want.  If the simulator is to publish after every step, it should publish after _every_ step, not just the ones that hit the max step size.

sherm1, 2016-12-05:
It looks as though the current event-detecting algorithm doesn't yet handle simulataneous events correctly. When update & publish occur at the same time (to be determined unambiguously by the integrator), the update should be performed followed by the publish call. This will be a good warm up for zero crossing-triggered events which also have the possibility of occurring simultaneously with update and publish. Looking at you, @edrumwri!

edrumwri, 2016-12-05:
Gotcha

david-german-tri, 2016-12-05:
OK, #4372 will give you publishes at `0.001, 0.002, 0.0025, 0.0035, 0.0045, 0.005, 0.006..`.  It will also enable you to register your own Publish frequency in SignalLogger with `DeclarePublishPeriodSec`, and turn off per-timestep publishing, if you strictly want 400Hz publishes.

david-german-tri, 2016-12-05:
> It looks as though the current event-detecting algorithm doesn't yet handle simulataneous events correctly. When update & publish occur at the same time (to be determined unambiguously by the integrator), the update should be performed followed by the publish call.

`System<T>::CalcNextUpdateTime` may return an update time that invokes discrete updates, publishes, or both, but it can't return update and publish events that have different times.  So, the logic to distinguish update and publish times in the Integrator was redundant and #4372 proposes to remove it.

david-german-tri, 2016-12-05:
(Ignore my comment above, see discussion on #4372)

RussTedrake, 2016-12-05:
Thanks David.  That at least makes sense.  But I would think that if there is no continuous state in the system then the integrator should not do any updates in between the registered discrete events.   Right?

david-german-tri, 2016-12-05:
> But I would think that if there is no continuous state in the system then the integrator should not do any updates in between the registered discrete events. Right?

That sounds reasonable to me, but the Simulator doesn't do any such check for continuous state right now.  I think that's a separate PR from #4372, and one I'm inclined to leave to @sherm1 and/or @edrumwri if that's OK?

PR #4372 (2016-12-05): Make publish-every-timestep configurable in the Simulator.
This PR does four closely related things:

1. Adds the ability to configure the Simulator to publish every timestep, or only when the System under simulation requests to publish.

2. When publish-every-timestep is on, fixes a bug so that we actually publish on every timestep, not just the timesteps that end due to hitting the max sample time.

3. Checks whether the System under publish has requested a discrete sample because it wants to publish, do a difference update, or both, and conveys that information to the integrator.

4. Adds s [... cut]
closed 2016-12-05 by david-german-tri