RobotLocomotion/drake #4361: std::uniform_real_distribution produces different numbers on mac clang vs ubuntu

RussTedrake, 2016-12-04:
Based on quick testing in #4358, it appears that the uniform distribution produces consistent results across the platforms, but that I'm seeing different results from std::uniform_real_distribution on mac and linux (the linux builds are all consistent with each other though).  

That would be consistent with the observation reported here:
http://stackoverflow.com/questions/34903356/c11-random-number-distributions-are-not-consistent-across-platforms-what-al

I wanted to open this issue for two reasons to document this finding, but also to discuss.  How much do we value getting consistent results here?  We could implement our own distribution (as in the Box-Muller in the post above), but presumably at some cost in terms of performance (statistical and cpu time).   What do folks think?

jwnimmer-tri, 2016-12-05:
I guess the way I think about it is the best way to minimize effort to reach the end goal.  Having platform-varying behavior is probably a meaningful cost, but so is writing and/or optimizing our own normal distribution.

Would it make sense to write the code in terms of a Drake normal distribution, but for now have it call the system C++11 normal distribution?  If that mostly works, it seems like an easy way to make progress but leave a seam open to insert a custom implementation.

But also, the code for `normal_distribution::operator()` is actually pretty terse, so maybe its cheap to in-tree our own anyway, and we should do that from the start.

liangfok, 2016-12-07:
Per the manipulation subgroup meeting this morning, I just wanted to clarify that the differences in `std::normal_distribution` are likely due to differences in the implementations of the standard C++ libraries on OSX vs. Ubuntu and not due to `clang` vs. `gcc`.

_(The above statement is a conjecture that needs to be verified.)_

RussTedrake, 2018-09-23:
I think we understand that the results are inconsistent across platforms, and are going to live with it.

jwnimmer-tri, 2023-07-27:
The pain of differing platforms has been increasing.  I'll probably try to implement some of the fixes discussed above.

jwnimmer-tri, 2024-02-21:
Most acutely, probably the `std::uniform_real_distribution` is more important to make cross-platform bit exact (vs the normal distribution).

Anyway if we're just going to vendor the Boost real number distributions then we might as well do all of them at once.