RobotLocomotion/drake #4371: DiagramBuilder's ExportInput() and ExportOutput() Methods Do Not Provide Port Number Information

liangfok, 2016-12-05:
# Problem Definition

When constructing a `Diagram` that exports more than one input and output ports, it is often necessary to know the index of a particular port. This is currently not possible because `drake::systems::DiagramBuilder::ExportInput()` and `drake::systems::DiagramBuilder::ExportOutput()` do not tell the calling code the indices of the ports that were just exported.

In #4296, the `PidControlledSystem` was modified to provide two accessors: `get_control_input_port()` and `get_state_input_port()`. The implementation of these methods hard-code the indices 0 and 1, respectively. This is brittle against the order in which these input ports were declared, which occurs [here](https://github.com/RobotLocomotion/drake/blob/9415dbdf0632ce7b4af08aa54f1a426b4fe580e1/drake/systems/controllers/pid_controlled_system.cc#L117).

# Suggested Solution

Modify `drake::systems::DiagramBuilder::ExportInput()` to return the index of the port that was just declared. This port index can then be used by `PidControlledSystem` to avoid hard-coding the port indices in `get_control_input_port()` and `get_state_input_port()`. Similarly, `drake::systems::DiagramBuilder::ExportOutput()` should be modified in the same manner and for the same purpose.

david-german-tri, 2016-12-05:
Good point!  The fix SGTM, feel free to PR it (or I'll get to it eventually)

liangfok, 2016-12-05:
OK thanks. Will do.

PR #5397 (2017-03-03): add return value to export input / output in diagram builder.
add return index to ExportInput / ExportOutput in diagram builder.

Fixes #4371

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5397)
closed 2017-03-03 by david-german-tri