RobotLocomotion/drake #4377: Design discussion: Sample Times in System 2.0

RussTedrake, 2016-12-06:
See [this google doc](https://docs.google.com/document/d/1NRg5E1ziZ27nnOCppZ7m-E3lXFE6a84IN-QCzlVJTo4) for my current understanding.  Let's iterate and implement.

sherm1, 2016-12-07:
How I think about discrete events in general: there are *triggers* and *event handlers*. In principal any kind of trigger can can invoke any kind of event handler. We currently distinguish three kinds of event handlers:
- "publish" events (can't affect the trajectory)
- "discrete" events (can only modify numerical "difference" variables xd)
- "unrestricted" events (can modify anything in the Context except time)

Typical triggers:
- successful completion of an integration step (trajectory advancement)
- periodic or specified time reached
- zero crossing of a state-dependent witness (guard) function
- arrival of a message or user input
- simulation initiation or termination

@edrumwri is currently implementing unrestricted events and came up with that term which I think is very accurate.

We will not necessarily implement the full cross product of (triggers X handlers); as long as we have (triggers X unrestricted) we can do anything.

david-german-tri, 2016-12-07:
I'm generally on board with @sherm1's framework.  I do hope there are intermediate waypoints that will let us update abstract state on a time trigger, though!  There's a lot of demand for that capability.

david-german-tri, 2016-12-07:
(And from an implementation standpoint, it would be pretty trivial to add to EvalDifferenceUpdates)

sherm1, 2016-12-08:
> I do hope there are intermediate waypoints that will let us update abstract state on a time trigger, though! There's a lot of demand for that capability.

Glad you said that! See @edrumwri's PR #4382.

david-german-tri, 2017-04-13:
Closing as fixed, as the result of many PRs.   As @sherm1 mentioned above, #4382 was especially significant.

closed 2017-04-13 by david-german-tri