RobotLocomotion/drake #4388: Enumerate Geometry dependencies

SeanCurtis-TRI, 2016-12-06:
GeometryWorld (or whatever it will eventually be called) is coming.  In short, this is the encapsulation with all things geometric; the data and operations on it.  This will allow us to integrate disparate systems which share underlying geometry (e.g., the static world.)  It will also allow us to simulate sensors on worlds where moving geometry is being driven by different systems (e.g., a dynamics-driven ego car vs. parametric driving models.)  They use different logic to move, but, at the end of the day, the end result is posed geometry in a common world.

While a fair amount of thinking has gone into this already, it is by no means complete. One of the most important aspects of this is making sure that we don't have any geometry-dependent processes left hung out to dry (or find that our design needs to be completely re-vamped just to handle *current* capability.)  

This issue is about capturing those downstream dependencies.  So, if you know of dependencies on geometry or geometric queries, let's gather them here.  I've primed the pump with a few cases I know.  This initial list is by no means exhaustive, but we want it to grow to be.  Feel free to add use cases that are not currently implemented but are coming -- please label them accordingly (see the `LidarSensor` below as an example.)

Dependencies
-------------------
  - [ ] `DrakeVisualizer`
    - [ ] Dispatch an *ordered* list of geometry the visualizer should load at initialization time.
    - [ ] At each time step, dispatch an ordered list of poses (which should map to the initial list of geometry).
  - [ ] `RigidBodyPlant`
    - [ ] Determine which of the `RigidBodyTree`'s collision elements are in contact to compute contact forces.
  - [ ] `LidarSensor` (currently only System 1)
    - [ ] Cast a set of rays on the scene to determine intersections relative to the ray origin.

david-german-tri, 2016-12-06:
As we've previously discussed f2f, I think renderers generally (including but not limited to `DrakeVisualizer`) just need to be able to obtain a pose (and eventually derivatives of pose) per object/model-instance.

jwnimmer-tri, 2016-12-06:
A slightly different use case is for something like an oracular object-level (aka target-level) sensor.  For that System, it will take as input the poses (and eventually derivatives of pose) of geometry, and produce a output that represents the body-relative position of (and eventually derivatives of position) the geometry with respect to the sensor's pose.  However, I believe that any solution that satisfies a visualizers needs would also satisfy this need.

SeanCurtis-TRI, 2016-12-06:
@jwnimmer-tri This case doesn't seem sufficiently detailed.  It implies that the sensor has access to the geometry (and the intent of `GeometryWorld` is that such is not the case.)  Secondly, the scenario you've described suggests the point is to "position" geometry.  Surely, that's the first step of an actual query?  Once positioned, then what?  I suspect that the case you have formulating in your mind *will* be resolved with what we intend for `GeometryWorld`.  I'd like to flesh it out to make sure.

SeanCurtis-TRI, 2019-05-13:
This served its purpose in informing the initial design. A lot of water has flowed under the bridge since then and any information about missing APIs is now directly coming from current events. As such, this issue no longer provides value.

closed 2019-05-13 by SeanCurtis-TRI