RobotLocomotion/drake #4397: On-the-fly mesh triangulation

SeanCurtis-TRI, 2016-12-07:
Problem
======

Most geometric algorithms which operate on meshes require triangulated meshes.  Our robot models may or may not be triangulated.  Rather than manipulating all of the data to make sure it is triangulated, we can introduce a mechanism by which mesh definitions can be triangulated in memory so it is always compatible with triangle mesh algorithms.

Why should I care?
------------------------

This is more than simply optimization.  Many of these algorithms don't just *run better* with triangles, they *only* run with triangles.  For example, right now, the underlying collision engine (bullet) requires meshes to be triangulated.  If the mesh is *not* triangulated, we replace the mesh with its convex hull (which is only a function of the vertices and *not* the topology).  If, for example, the mesh in question is a definition of terrain with rolling hills, all of the valleys will be filled up when creating the convex hull. 

Pros and Cons
-------------------

Pros
- No requirement for pre-massaging the input geometric data to make it *technically* compliant (i.e., making sure it's all triangles.)

Cons
- If polygons are distinctly not planar, automated triangulation may produce undesriable results (or, at the very least, less desirable than what we would get with a human in the loop.)
- We pay a run time cost *everytime* for a problem that could be solved once in the input files.
- There is a disconnect between what is in memory and what is in file -- if someone, god forbid, had a sense of triangular indices, those would be invalidated.

Alternatives
---------------- 

We simply fail hard when finding and incompatible mesh, report a message to the user that they need to bring their inputs up to spec.

liangfok, 2016-12-07:
I'm very nervous about having what's in memory differ from what's in the file since it may complicate debugging. Thus, I vote for failing hard but providing a tool (or pointers instructions using, for example, MeshLab) that allows users to convert their non-triangulated meshes into triangulated meshes.

SeanCurtis-TRI, 2017-01-13:
@liangfok The "fail hard" approach led to a *mountain* of test failures.  Hence the fall back to the convex hull.  So, if we were to go with the fail hard, we'd have to go through all of the current meshes and triangulate them.

Alternatively, the difference between the mesh in memory and on disk is slight.  Vertex position doesn't move, we only subdivide faces that are too large.  If we put out a log message indicating that it was being triangulated, would that alleviate the concerns in the differences?

closed 2017-01-24 by SeanCurtis-TRI