RobotLocomotion/drake #4398: Order the state for a floating-base body as [quaternion; translation; angular_velocity; linear_velocity] when using kQuaternion

hongkai-dai, 2016-12-07:
Moved the discussion from #4379. Currently the order is
```
[translation; quaternion; angular_velocity; linear_velocity]
```
This can cause confusing to users, since it puts translational part first in the generalized position, but switch the order in the generalized velocity, that the angular velocity comes first.

siyuanfeng-tri, 2016-12-08:
for euler angles, i believe the current layout is pos, rpy, posd, rpyd. 
so we are sticking to "everything rotation comes first"?

hongkai-dai, 2016-12-08:
@siyuanfeng-tri I think only the drone people are using euler angles. @peteflorence @psiorx , do you have a preference to the ordering of the state for your drone?

sherm1, 2016-12-08:
> for euler angles, i believe the current layout is pos, rpy, posd, rpyd. so we are sticking to "everything rotation comes first"?

At least the RPY floating joint is self-consistent between coordinates and velocities! I do think we should be consistent and always use orientation followed by translation for q and v. As a practical matter, the first step ought to be fixing the API so that user code doesn't have to be so brittle with respect to coordinate ordering. Then find and remove the brittle parts, then finally make the ordering changes hopefully without breaking anything.

siyuanfeng-tri, 2016-12-08:
i can see this creating relatively large confusions to the current user base if we want to fix it.. (i got bitten by this a couple times already). on the other hand, i am strongly in favor of moving towards a consistent convention.

siyuanfeng-tri, 2016-12-08:
@hongkai-dai we are as well unfortunately, i am not exactly sure for what reason. probably just history.

siyuanfeng-tri, 2016-12-08:
i would guess most people want [pos, rotation]

sherm1, 2016-12-09:
/cc @mitiguy

closed 2019-01-25 by hongkai-dai