RobotLocomotion/drake #4400: Systems implement EvalTimeDerivatives but DoEvalDifferenceUpdates and DoPublish

RussTedrake, 2016-12-07:
... because e.g. there is a little bit of wrapper logic in EvalDifferenceUpdates in System.

Can we make the naming more consistent at the (Leaf)System implementation level?  I would prefer, e.g., for systems to implement EvalDifferenceUpdates, and to change EvalDifferenceUpdates at the System level to something more like EvalDifferenceUpdatesWrapper.

pinging @david-german-tri , @sherm1 for their thoughts.

sherm1, 2016-12-07:
I'm planning to make a naming-consistency pass for the caching project in #4364. We have been using the NVI pattern inconsistently so far but should do it uniformly. That always involves a non-virtual public interface like `Publish()` that can do some common error checking, instrumentation and the like and then invoke a protected virtual like `DoPublish()`. I've never found a particularly nice convention for naming the virtual, but I think the current convention of prefixing the name with `Do` is as good as any. I think the nicer name should be the non-virtual though since that is what end users see in doxygen. 

The uniform pattern I'd like to see for cached values is 
```c++
public:
  const Thing& EvalThing(const Context&) const; // non virtual, invokes CalcThing() if necessary
  void CalcThing(const Context&, Thing*) const; // unconditional computation of Thing into arg
protected:
  virtual void DoCalcThing(const Context&, Thing*) const;
```

`EvalThing()` looks like this:
``` c++
const Thing& EvalThing(const Context& context) const { 
  auto& entry = context.get_cache_entry<Thing>(thing_cache_ticket_);
  if (!entry.is_valid())
    CalcThing(context, entry.get_mutable_value());
  return entry.get_value();
}
```

RussTedrake, 2016-12-07:
As an author of systems, I'd be fine if we implemented a bunch of Do___ methods.  And as an author of algorithms, we call a bunch of ___ methods.  (I think that's what we're saying).  My request is primarily for consistency!

PR #4520 (2016-12-21): First round of name changes for caching implementation
Renames methods to consistently use the Eval/Calc/DoCalc naming scheme for cacheable computations. This is intended to be a no-op change needed as a first step to making the Eval methods use the cache, [like this](https://github.com/RobotLocomotion/drake/issues/3155#issuecomment-240258256). For where this is headed see #4364. Fixes #4400.

I am hoping to implement this in several easy(ish)-to-review passes. To that end this pass is not intended as the final destination for many of the changes it includes. This provides consistent naming so [... cut]
closed 2016-12-21 by sherm1