RobotLocomotion/drake #4415: Support symbolic expression constraints in MathematicalProgram.

hongkai-dai, 2016-12-07:
<h1> Goal </h1>
To support symbolic constraints/costs in conic programming problems.

Example (how user will write the code):
```
MathematicalProgram prog;
auto x = prog.AddSymbolicContinuousVariables<2>({"x0", "x1"});
// Add constraint 0 <= x(0) + 2*x(1) <= 1
prog.AddLinearInequalityConstraint(x(0) + 2*x(1), 0, 1);
```

<h1> Proposed Solution </h1>
Based on the discussion between @soonho-tri and I, we decide to take the following steps.

1. Derive `symbolic::DecisionVariable` from `symbolic::Variable`, this `symbolic::DecisionVariable` contains a `solvers::DecisionVariableScalar` object. Since we can already construct `symbolic::Expression` from `symbolic::Variable`, the returned object `x` in the above example `auto x = prog.AddSymbolicContinuousVariables<2>({"x0", "x1"});` is a `Eigen::Matrix<symbolic::Expression, 2, 1>` object. 
   Currently we support the mathematical operations, such as multiplication, addition, plus, etc., between `symbolic::Expression`. So `x(0) + 2*x(1)` in the aboved example will be a `symbolic::Expression` object.
2. We will add new methods to `MathematicalProgram`, such as
   ```
   void AddLinearInequalityConstraint(const symbolic::Expression& expression, double lb, double ub);
   void AddLinearInequalityConstraint(const drake::VectorX<symbolic::Expression>& expression, const Eigen::Ref<const Eigen::VectorXd> lb, const Eigen::Ref<const Eigen::VectorXd> ub);
   void AddLinearInequalityConstraint(const drake::MatrixX<symbolic::Expression>& expression, const Eigen::Ref<const Eigen::MatrixXd> lb, const Eigen::Ref<const Eigen::MatrixXd> ub);
   ```
   These methods allow us to specify constraints `lb <= expression <= ub`. Inside these methods, they first extract the `DecisionVariableScalar` objects from the expression, then determine if the expression is affine w.r.t the decision variables. If it is, then return the double matrices `A`, `b` such that `expression = A*x + b`. Finally it calls the existing methods such as `AddLinearInequalityConstraint(A, b, lb, ub, {x})`.

<h2> Extensions </h2>

1. **Sum-of-squares polynomial**. Inside a sum-of-squares polynomial, such as `a*x^2 + b*x + c`, it contains two types of variables: (1) indeterminates `x`. (2) decision variables `a, b, c`. If we want to impose the constraint `a*x^2+b*x+c >=0 for all x`, we need to decompose this polynomial to the product of indeterminates and decision variables 
   ```
   a*x^2+b*x+c = [x]^T * [a   b/2] * [x]
                 [1]     [b/2   c]   [1]
   ```
   Namely, we need to differentiate between indeterminates and decision variables. @soonho-tri and I plan to add another class `symbolic::Indeterminate`, also derived from `symbolic::Variable`. And the class hierarchy would look like
   ```
                   symbolic::Variable
        /                               \
      /                                   \
   symbolic::DecisionVariable      symbolic::Indeterminate
   ```
   Thus we can write the program as
   ```
   MathematicalProgram prog;
   Vector3<symbolic::Expression> abc = prog.AddSymbolicContinuousVariable<3>({"a", "b", "c"});
   Vector1<symbolic::Expression> x = prog.AddIneterminate<1>({"x"});
   symbolic::Expression sos_poly = abc(0)*x*x + abc(1)*x + abc(2);
   ```
   Then we will need to write function like `decompose` or `recompose` for `symbolic::Expression` class, so as to decompose the polynomial into monomials and decision variables. @jadecastro and I should talk about this.
2. **Positive Semidefinite Constraint, Second-order cone constraint** The proposal above supports adding linear constraints using symbolic expressions. Ideally I wish we can add other conic constraints, such as positive-semidefinite constraints, and second-order cone constraints using symbolic expressions. So that I can write program like the following
   ```
   // This program finds a Lyapunov function for a linear system xdot = A*x + b
   MathematicalProgram prog;
   Eigen::Matrix3d A = Blah;
   auto P = prog.AddSymbolicSymmetricVariable<3>("P");
   prog.AddPositiveSemidefiniteConstraint(P);                      // Add psd constraint for a matrix with decision variables.
   prog.AddPositiveSemidefiniteConstraint(-(A'*P + P*A));   // Add psd constraint for a matrix containing symbolic expressions.
   ```
   Thus, we should also overload `AddPositiveSemidefiniteConstraint` and `AddLorentzConeConstraint` to accept `Eigen::Matrix<symbolic::Expression,...>` objects.

sherm1, 2016-12-07:
BTW if you are looking for a nice operator in C++ to mean transpose in symbolic expressions, you can't overload `'` like in Matlab, but consider `operator~()`. Then the Matlab `A'*P + P*A` looks like `~A*P + P*A` which is pretty close once you get used to it. It also has the right precedence and has kind of a "transposey" look to its wiggly shape.

hongkai-dai, 2016-12-07:
@sherm, thanks for the suggestion. I think our `A` matrix or `P` matrix are both Eigen::Matrix, we just put different scalar types in the Eigen::Matrix. So we can use the `transpose()` function in Eigen. I was sloppy when I wrote that code, it should be
```c++
prog.AddPositiveSemidefiniteConstraint(-(A.transpose()*P + P*A));
```

edrumwri, 2016-12-07:
I vote against overloading that operator. I pity the poor user/maintainer
that has to figure out what it does.

jadecastro, 2016-12-07:
SGTM. I could imagine that it would be somewhat trivial to create a special case encompassing equality constraints, for convenience to the user, and perhaps allow the user to specify an optional tolerance, like so: `AddLinearInequalityConstraint(symbolic_expr, epsilon)`.

Out of curiosity, could you say something like `AddLinearInequalityConstraint(A, b, lb, ub)`, where `A` is some `m`-by-`n` matrix and `b` is `m`-by-1?  I didn't see an intermixing of `MatrixX` with `VectorX`, which is why I'm asking.

hongkai-dai, 2016-12-07:
@jadecastro you are right, I used the wrong function signature. I just updated my proposal above. We should use `AddLinearConstraint(A, lb-b, ub-b, {x})`. Adding linear equality constraint with symbolic expression should be very much the same as adding linear inequality constraint. BTW, what is the tolerance `epsilon` used for a linear inequality constraint?

@soonho-tri To keep it consistent with the current implementation, shall we name the method as `AddLinearConstraint` instead of `AddLinearInequalityConstraint` when we add a linear inequality constraint `lb <= expression <= ub`?

jadecastro, 2016-12-07:
Sorry, I was off slightly with my function signature in my last post (I meant to say "equality" constraint). I've updated that -- see if that makes the difference. `epsilon` would relax the constraint symmetrically about zero. So, `-epsilon <= <expression> <= epsilon`.  You could just write out two expressions also, so it's really not a bummer if it's not there.

hongkai-dai, 2016-12-07:
@jadecastro I think since we already have `AddLinearConstraint` for linear inequality constraint, then `AddLinearEqualityConstraint` should really mean equality constraint. The reason why we want to make that distinction, is that linear equality constraint and linear inequality constraint can incur different solvers. If we have a quadratic program with only linear equality constraints, then we can find the analytical solution; with inequality constraints we cannot do that, but have to determine which inequality constraint is active. 

If you mean the solver might not satisfy the equality constraints exactly, but can be off by some numeric precision, then we can set the feasibility tolerance for each solver. snopt is default to be accurate to 1E-6, Mosek is default to 1E-8. We can tune these knobs.

jadecastro, 2016-12-07:
Thanks! I meant the former. In that case I agree we should get rid of the
epsilon.

soonho-tri, 2016-12-08:
> @soonho-tri To keep it consistent with the current implementation, shall we name the method as AddLinearConstraint instead of AddLinearInequalityConstraint when we add a linear inequality constraint lb <= expression <= ub?

No problem.

RussTedrake, 2016-12-08:
Sorry I'm late to the party.  I thought the original decision was to derive DecisionVariableScalar from Symbolic?

soonho-tri, 2016-12-08:
> I thought the original decision was to derive DecisionVariableScalar from Symbolic?

That was the original plan. But I realize that there is an important difference between `symbolic::Variable` and `solvers::DecisionVariableScalar`, which requires changes in the original plan.

 1.  `solvers::DecisionVariableScalar` is *mutable* mainly because it includes a value assignment (`double *`) inside.
 2. `symbolic::Variable` is immutable -- it has `const` members and its copy/move-assignment operators are deleted. It does not include a value assignment in it. Evaluation of symbolic expressions requires a mapping from a `symVar` to its value (i.e. `double`).

> The "AddSymbolicVariable" doesn't really make sense... and I'd love to be able to get symbolic expressions from operations on any variable.

I like this idea. @hongkai-dai also mentioned it last night in our discussion. Please note that this can be done without making `solvers::DecisionVariableScalar` a derived class of `symbolic::Variable`. We need operator overloadings over `solvers::DecisionVariableScalar` and (possibly) a constructor of `symbolic::Expression` directly taking `solvers::DecisionVariableScalar`. I'll do this soon.

soonho-tri, 2016-12-08:
@hongkai-dai, I'd love to have your opinion on my comment above regarding adding operator overloadings for `solvers::DecisionVariableScalar`. For example, we can have:

```c++
symbolic::Exprsesion operator+(const solvers::DecisionVariableScalar & v1, 
                               const solvers::DecisionVariableScalar & v2);
```

My current plan is to provide these overloading in the `drake/common/symbolic_expression.h` header so that these overloadings are enabled only if the symbolic header is included.

hongkai-dai, 2016-12-08:
Based on f2f discussion with @soonho-tri , @RussTedrake  and @sammy-tri , I updated the proposal on top. @soonho-tri 's suggestion is great, we should overload some operators for DecisionVariableScalar.

soonho-tri, 2016-12-28:
@hongkai-dai , When https://github.com/RobotLocomotion/drake/pull/4562 is merged, I'll continue working on this issue. Could you confirm that we still need the following methods in the item3?

```c++
void AddLinearConstraint(const symbolic::Expression& expression, double lb, double ub);
void AddLinearConstraint(const Eigen::Ref<const drake::VectorX<symbolic::Expression>> expression, const Eigen::Ref<const Eigen::VectorXd> lb, const Eigen::Ref<const Eigen::VectorXd> ub);
void AddLinearConstraint(const Eigen::Ref<const drake::MatrixX<symbolic::Expression>> expression, const Eigen::Ref<const Eigen::MatrixXd> lb, const Eigen::Ref<const Eigen::MatrixXd> ub);
```

hongkai-dai, 2016-12-28:
@soonho-tri , yes we still need this. I just replaced `DecisionVariableScalar` with `symbolic::Variable`, I will send a PR today.

soonho-tri, 2016-12-28:
I see.

> I just replaced DecisionVariableScalar with symbolic::Variable, I will send a PR today.

Sounds good!

soonho-tri, 2017-01-18:
@hongkai-dai , the proposed solution part is done. How about closing this one?

I believe that there is a separate issue https://github.com/RobotLocomotion/drake/issues/4421 for the first item of the extension part above. But I'm not sure that we have an open issue for the second item of the extension part.

closed 2017-01-18 by hongkai-dai