RobotLocomotion/drake #4424: model.config and model:// are Unsupported

liangfok, 2016-12-08:
Many SDF files and associated resources (e.g., mesh files) are stored in directories with the following structure:

1. The directory name is the name of the model.
2. The directory contains a `model.cfg` file that contains details about the SDF file(s).
3. The SDF file references resources using `model://` syntax.

Examples: http://models.gazebosim.org/

If we want to use these models, we should ideally update Drake to support the above format so we don't have to fork them.

jwnimmer-tri, 2016-12-22:
We don't need this _yet_ in any of our exiting models.  We can reconsider if we do need it later.

closed 2016-12-22 by jwnimmer-tri

liangfok, 2017-03-08:
This issue is block on #4337. It does not make sense to implement this in the current parsers, which will soon be replaced by `libsdformat`. (Maybe the new parser has built-in support for this? Not sure.)