RobotLocomotion/drake #4427: Simulator<AutoDiffXd>::StepTo assertion

jwnimmer-tri, 2016-12-08:
In #4422, we discovered that `Simulator<AutoDiffXd>::StepTo` fails an assertion.  We should enhance the `Simulator<AutoDiffXd>` unit testing to call StepTo, observe that the assertion trips, and then repair the code to work (and possibly adding addition tests).

sherm1, 2016-12-08:
/cc @edrumwri

sherm1, 2016-12-08:
Also likely upgrade that assertion to a DRAKE_DEMAND.

sherm1, 2016-12-08:
I traced this to some code in LeafSystem that looked like this:
```c++
    T min_time =
        std::numeric_limits<typename Eigen::NumTraits<T>::Real>::infinity();
```
where T was an AutoDiffScalar. The problem is that `::Real` here doesn't mean "underlying floating point type", rather it means "real part" vs. "imaginary part". Since T wasn't complex, Real is simply T. `numeric_limits` has no instantiation for that so uses its default which is to return a `T()`, garbage in this case.

Replacing `::Real` with `::Literal` (as in type of a literal for this type, say `3.14`) fixes the problem. PR coming.

PR #4431 (2016-12-09): Use NumTraits::Literal rather than ::Real
Replaces two uses of `Eigen::NumTraits<T>::Real with `NumTraits<T>::Literal` where the intention was to get the underlying floating point type. `::Real` actually means "real part" (vs. "imaginary part"). There are several other uses of `::Real` in Drake that appear to be using it correctly.

Enables the AutoDiff test that failed in PR #4422. Upgrades the assertion that failed to a demand. 

Fixes #4427.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4431)
commit 7bf8ac9c (2016-12-09): Replace NumTraits::Real with NumTraits::Literal. Fixes #4427.
closed 2016-12-09 by david-german-tri