RobotLocomotion/drake #4434: SteeringCommandDriver.py Throttle Can Only Increase

liangfok, 2016-12-09:
# Problem Definition

When running `drake-distro/drake/automotive/steering_command_driver.py`, the throttle value can only be increased, not decreased.

liangfok, 2016-12-09:
@shensquared do you have any insights into this?

shensquared, 2016-12-09:
Yes, I think it used to be binary, pressing the up arrow sends throttle value 1, and releasing the up arrow sends throttle value 0. My PR changes pressing the up arrow to gradually increase the throttle, up to ~26, but broke the releasing up arrow key part: the throttle now keeps the value when the key was released. I'll fix it, will set releasing the up arrow to decrease the throttle to 0 at the same rate that pressing the key increases it.

PR #4458 (2016-12-15): fixes SteeringCommandDriver.py throttle not being able to decrease
This fixes #4434, and adds the option of keeping the current throttle/brake. It also adds keyboard control instruction for better usability. 

+@liangfok for feature review, thanks.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4458)
closed 2016-12-15 by jwnimmer-tri