RobotLocomotion/drake #4442: Parser Methods Need to be Simplified

liangfok, 2016-12-09:
# Problem Definition

Per @jwnimmer-tri's review comments in #4363, there's currently a bajillion parser functions in the parsers and some have 70+-character function names. Let's simplify things.

# Proposed Solution

Let's provide only four methods:

1. Add from file.
2. Add from file while searching in ROS packages.
3. Add from string.
4. Add from string while searching in ROS packages.

If we can standardize on a way to get the ROS package map (#1471), we can eliminate the two "searching in ROS packages" above leaving only 2 methods.

jwnimmer-tri, 2016-12-09:
Perhaps making the loader a class would make it even terser.  "Make a loader; tell it the package name (and other loading settings); call load on a filename".

liangfok, 2016-12-09:
Good point. TTBOMK, there are now three valid reasons to use a `class` instead of stand-alone methods:

1. To avoid a bunch of methods that all operate on the same output parameter object.
2. To avoid stand-alone methods that have super long names.
3. To maintain persistent state across method calls through member variables.

amcastro-tri, 2018-06-11:
This is one of the motivations for the migration to use `sdformat` for parsing, it does provide this parser class as here described. It still is in the works, but @nkoenig is working on it.
I will close this since @nkoenig already addressed exactly what this PR describes by introducing more semantically meaningful objects to `sdf::`.

closed 2018-06-11 by amcastro-tri