RobotLocomotion/drake #4445: Drake Contains Nested Packages That are Invisible to Other ROS Nodes.

liangfok, 2016-12-09:
# Problem Definition

Drake currently contains numerous packages that are nested within Drake's top-level package, which is `drake-distro/`:

```
$ cd drake-distro
$ find . -name "package.xml"
./build/install/share/octomap/package.xml
./drake/examples/Atlas/package.xml
./drake/examples/IRB140/package.xml
./drake/examples/Valkyrie/package.xml
./drake/multibody/rigid_body_system1/test/package.xml
./drake/multibody/test/package_map_test/package_map_test_packages/package_map_test_package_a/package.xml
./drake/multibody/test/package_map_test/package_map_test_packages/package_map_test_package_b/package.xml
./drake/multibody/test/package_map_test/package_map_test_packages/package_map_test_package_set/package_map_test_package_c/package.xml
./drake/multibody/test/package_map_test/package_map_test_packages/package_map_test_package_set/package_map_test_package_d/package.xml
./externals/octomap/dynamicEDT3D/package.xml
./externals/octomap/octomap/package.xml
./externals/octomap/octovis/package.xml
./package.xml
./ros/drake_examples_toyota_hsrb/package.xml
./ros/drake_ros_common/package.xml
./ros/drake_ros_common_test/package.xml
./ros/drake_ros_integration/package.xml
./ros/drake_ros_systems/package.xml
./ros/hsrb_description/package.xml
```

While these nested packages are visible to Drake's parsers, they are _not_ visible to other ROS nodes since ROS does not support nested packages. This can be problematic when Drake is collaborating with other ROS nodes since Drake may refer to a package that is not visible to the other ROS nodes.

The aforementioned package visibility differences may result in Drake using a different package than other ROS nodes running alongside it. For example, Drake might one day use its internal `Valkyrie` package (located in `drake-distro/drake/examples/Valkyrie`). Because this package is not visible to the other ROS nodes, the other ROS nodes might end up using the official [valkyrie_description](https://gitlab.com/nasa-jsc-robotics/val_description/tree/master) package. Keeping the two models in sync may become a maintenance problem.

# Suggested Solution

I believe the ideal solution is to un-nest the nested packages within `drake-distro`. Doing this would require modifying Drake's build system so that `drake-distro/package.xml` can be moved into `drake-distro/drake/package.xml` (The necessary modifications to Drake's build system were described by @jhoare here: https://github.com/RobotLocomotion/drake/issues/2116#issuecomment-236264384.). Then, the existing nested packages can be moved to be at the level of `drake/distro`. Since they are no longer nested, they would become visible to all ROS nodes. Since they are still within `drake-distro`, they should still remain visible to Drake's parsers.

jwnimmer-tri, 2016-12-10:
I would prefer not perverting Drake's source tree layout to satisfy ROS weirdness.  Could we instead (offer an option to) _install_ our files to a ROS-compatible layout?

liangfok, 2016-12-10:
I believe the above-proposed layout will allow Drake to comply with ROS wierdness without needing a separate ROS-specific install step. Specifically, here is the proposed layout:

```
drake-distro/drake/package.xml  -- the `drake` package, depends on `drake_externals`
drake-distro/drake_externals/package.xml -- the `drake_externals` package
drake-distro/ros/ -- contains subdirectories, each its own package, depends on `drake`
drake-distro/models/ -- each subdirectory is its own package
drake-distro/examples/ -- each subdirectory is its own package, depends on `drake`
```

jwnimmer-tri, 2016-12-10:
Moving stuff from `drake-distro/drake` back up into `drake-distro` is a big change, that we should be deliberative about.  Maybe we can fix ROS instead?

liangfok, 2016-12-10:
One way to make Drake compatible with ROS would be to use symbolic links that go from a sibling of `drake-distro` to the various nested-packages that are currently within `drake-distro/drake/`. This is similar to how we're currently using a symbolic link for `drake-distro/ros/`.

liangfok, 2017-04-12:
I'm going to close this issue since the way Drake + ROS is integrated will need to be revamped once #3129 is resolved.

closed 2017-04-12 by liangfok