RobotLocomotion/drake #4446: Simulator StepTo() status return needed

sherm1, 2016-12-09:
The Simulator's StepTo() method can return early for a variety of reasons. At times it is important for the caller to be able to ask what happened:
- reached the final time
- an event handler requested termination
- an event handler detected a condition that requires a high-level reinitialization (like adding/removing new objects from a simulation)

Most of the time no one is interested in the return status so it would be reasonable to implement this with a retained "termination condition" that can optionally be checked.

jwnimmer-tri, 2019-03-29:
@sherm1 or @edrumwri Rename this ticket to AdvanceTo instead of StepTo, now?

Or just resolve this as a duplicate of #4447 (as a checklist item within that work)?  It seems to me like only the "requested termination" status is relevant for now.

sherm1, 2019-03-29:
Renamed -- they are independent enough that I'd like to keep both issues.

sherm1, 2019-11-08:
The status return ([SimulatorStatus](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_simulator_status.html)) was added in PR #12213. Pending resolution of #4447, only the monitor() function's EventStatus return actually bubbles up to AdvanceTo(). But the status return is there now and ready to absorb more sources of EventStatus.

closed 2019-11-08 by sherm1