RobotLocomotion/drake #4448: Add SisoVectorSystem (and related)

jwnimmer-tri, 2016-12-10:
We can add even more System 2.0 sugar for the common case of `LeafSystem`s that have at most one input port, one output port, and optionally continuous-only state.

WIP at https://github.com/jwnimmer-tri/drake/tree/systems-single-base though I probably won't return to it for a while.

I have separate base classes for output-only and input-and-output systems.  I've tentatively named this single-input single-output systems (SisoSystem) in my branch, but we can name-wrangle later on.

jwnimmer-tri, 2016-12-10:
If anyone has an acute need for this, I'd be happy to give my branch to someone else to finish if you wish.

RussTedrake, 2016-12-10:
Here's a link (from @jwnimmer-tri) for folks to view the proposed API:
https://github.com/jwnimmer-tri/drake/blob/systems-single-base/drake/systems/framework/primitives/siso_vector_system.h#L109

@mposa -- this is what I was telling you about last night.  @jwnimmer-tri -- there is a lot of interest from the matlab users of drake that are considering the move to c++.  I think it will be a great thing to have.

jwnimmer-tri, 2016-12-10:
I've rebased to latest master API changes and added a bit more implementation.  Unclear how quickly I'll make progress to wrap this up, but I'll keep it on deck.

jwnimmer-tri, 2016-12-12:
From a survey of existing systems in-tree, there seem to be common cases worth adding sugar for:
 1. `SingleOutputVectorSource`: 0 input, 0 state, 1 output vector
 2. `SisoFeedthroughVectorSystem`: 1 input vector, 0 state, 1 output vector, direct-feedthrough
 3. `SisoNonfeedthroughVectorSystem`: 1 input vector, 1 continuous state vector, 1 output vector identical to continuous state, not direct-feedthrough

Uses of (1):
- ConstantVectorSource
- TrajectorySource
- TBD Possible unit-testing source that can be easily fixed to different values while testing

Uses of (2):
- Gain
- Passthrough
- RotaryEncoders
- GravityCompensator

Uses of (3):
- Integrator
- LuenbergerObserver
- NOT ZeroOrderHold (because output comes from _discrete_ state)

Do those three seem like useful base classes to distinguish?  Anyone with systems / controls background want to choose better names for me?

Also, if one of these is particularly more useful that the others, I can work to get it into master first.

RussTedrake, 2016-12-12:
The list of three looks good. 

- The first seems like the least important (the current implementations probably cover most of the use cases, and the full syntax is not so cumbersome)
- the emphasis on feedthrough is not the most important distinction between 2 and 3, so i would tweak the names.  The bigger distinction is whether they have any state.  Declaring feedthrough is also not much of a burden on the author.  I might suggest, instead:
  - SisoContinuousSystem
  - SisoStatelessSystem
And add
  - SisoDiscreteSystem

All would have 1 input, 1 output, and 0 or 1 state vector.  

If we want to match matlab drake, then the input would be available to all in EvalOutputs, and they declare directfeedthrough.  Removing that argument and making directfeedthrough false is a fair proposal, but might not cover all the cases.

RussTedrake, 2016-12-12:
More quick thoughts: 
- your wrapper could call is_direct_feedthrough itself and dispatch to EvalOutputs with or without the input available (or throw).

- you could templte eg SisoContinuousSystem on input type, state type, and output type.  Those types must derive from basicvector, and basicvector would be the default.  But you could handle the downcast for people. This would cover eg pendulum/acrobot/quadrotor and probably become the most important case to do first.

RussTedrake, 2016-12-12:
Even better would be to have a mandatory feedthrough constructor argument.

jwnimmer-tri, 2017-01-20:
I think the number of times that API changes to master have broken my WIP branch for this feature (lots!) is a good indication that this work is sorely needed.  I will prioritize and push some PRs next week.

jwnimmer-tri, 2017-02-01:
@RussTedrake I think we can make this work with only a single Siso base class.  I am currently working through some testing and porting using the following APIs offered by SisoVectorSystem for use by the concrete subclasses:
```c++
/// A base class that specializes LeafSystem for use with only a single input
/// port, and only a single output port.
///
/// @tparam T The vector element type, which must be a valid Eigen scalar.
template <typename T>
class SisoVectorSystem : public LeafSystem<T> {
...
 protected:
  /// Creates a system with one input port and one output port of the given
  /// sizes.  Does not declare any state -- subclasses may optionally declare
  /// continuous or discrete state, but not both.
  SisoVectorSystem(int input_size, int output_size) {
    DRAKE_DEMAND(input_size > 0);
    DRAKE_DEMAND(output_size > 0);
    this->DeclareInputPort(kVectorValued, input_size);
    this->DeclareOutputPort(kVectorValued, output_size);
  }

  /// Performs the same logical operation as DoEvalOutput, but on VectorBlocks
  /// that have already been consistency-checked with the sizes declared at
  /// construction-time.
  ///
  /// The @p state will be either empty, the continuous state, or the discrete
  /// state, depending on which (or none) was declared at construction-time.
  virtual void DoCalcVectorOutput(
      const Context<T>& context,
      const Eigen::VectorBlock<const VectorX<T>>& input,
      const Eigen::VectorBlock<const VectorX<T>>& state,
      Eigen::VectorBlock<VectorX<T>>* output) const = 0;

  /// Performs the same logical operation as EvalTimeDerivatives, but on
  /// VectorBlocks that have already been consistency-checked with the sizes
  /// declared at construction-time.
  ///
  /// The @p state will be either empty or the continuous state, depending on
  /// whether continuous state was declared at construction-time.
  ///
  /// By default, this function does nothing if the @p derivatives are empty,
  /// and aborts otherwise.
  virtual void DoCalcVectorTimeDerivatives(
      const Context<T>& context,
      const Eigen::VectorBlock<const VectorX<T>>& input,
      const Eigen::VectorBlock<const VectorX<T>>& state,
      Eigen::VectorBlock<VectorX<T>>* derivatives) const {
    DRAKE_DEMAND(derivatives->size() == 0);
  }

  /// Performs the same logical operation as DoCalcDiscreteVariableUpdates, but
  /// on VectorBlocks that have already been consistency-checked with the sizes
  /// declared at construction-time.
  ///
  /// The @p state will be either empty or the discrete state, depending on
  /// whether discrete state was declared at construction-time.
  ///
  /// By default, this function does nothing if the @p discrete_updates are
  /// empty, and aborts otherwise.
  virtual void DoCalcVectorDiscreteVariableUpdates(
      const Context<T>& context,
      const Eigen::VectorBlock<const VectorX<T>>& input,
      const Eigen::VectorBlock<const VectorX<T>>& state,
      Eigen::VectorBlock<VectorX<T>>* discreate_updates) const {
    DRAKE_DEMAND(discrete_updates->size() == 0);
  }
};
```

Subclasses can declare `xc` or `xd` in their constructor, with no need to feed that information through to the `SisoVectorSystem` constructor.  They must override (or else fail the `DEMAND` at runtime) the derivative/update methods iff they declare the corresponding type of state.  The `...Output` method magically is provided either `xc` or `xd` depending on whichever kind was declared.  Using both `xc` and `xd` at once is not allowed (fails-fast at `Context`-creation time).

One uncertainty... I _could_ provide a default `...Output` implementation that simply outputs the state, which seems to be a common choice.  On the other hand, overriding that method is relatively brief, so perhaps not worth the risk of a user forgetting to specify the output?  I think I come down on ease-of-use for this onramp-like class and so outputting the state by default for less syntax, but I will defer to your judgment.

RussTedrake, 2017-02-02:
I like it.  I don't think you should provide that default output -- as you say, it's trivial to implement and we should ask the user to do it.  
n.b. update the doc from e.g. `EvalOutput` -> `CalcOutput`

jwnimmer-tri, 2017-02-02:
Awesome.  /CC @david-german-tri or @sherm1 if you want to offer any API concerns.

On my branch I have this implemented, and all relevant `primitives` and some `sensors` ported.  I'll shoot for a first PR by late next week.

Reading the API paste again, I wonder if we should offer a typedef for `const Eigen::VectorBlock<const VectorX<T>>` and `Eigen::VectorBlock<VectorX<T>>`.  The signatures are a bit wordy without one, but perhaps the typedef would obscure the underlying type too much.

sherm1, 2017-02-03:
Looks good. A few comments:
- Could the arguments be `Eigen::Ref` rather than `Eigen::VectorBlock`s? I believe that would still be happy with VectorBlocks but would allow more general arguments also. Could be good for testing at least.
- I think a `VectorBlockX<T>` would be a nice addition to our alias collection and wouldn't obscure anything.

david-german-tri, 2017-02-03:
:lgtm: cool!

jwnimmer-tri, 2017-02-06:
Interesting.  So we have two options.  (Assume for each that we could add typedefs for shorter syntax.)

Option 1:
```c++
virtual void DoCalcVectorOutput(
      const Context<T>& context,
      const Eigen::VectorBlock<const VectorX<T>>& input,
      const Eigen::VectorBlock<const VectorX<T>>& state,
      Eigen::VectorBlock<VectorX<T>>* output) const = 0;
```

Option 2:
```c++
virtual void DoCalcVectorOutput(
      const Context<T>& context,
      const Eigen::Ref<const VectorX<T>>& input,
      const Eigen::Ref<const VectorX<T>>& state,
      Eigen::Ref<VectorX<T>> output) const = 0;
```

I will work on a pro / con list for the two choices.

jwnimmer-tri, 2017-02-06:
1. Both types disallow resizing across `operator=` and direct `resize` calls (for both, it fails at runtime in Debug mode only).
2. Both types have the weird double-const pattern that will have to bleed into any typedefs we offer (because both options use wrapped-reference semantics).
3. Both types are (I think) cheap to copy.
4. In the two options above, I've had `output` sometimes a pointer or sometimes a (wrapped-reference) value.  I suspect that choice is uncorrelated with `Block` vs `Ref`, so will treat it separately.

The thing I haven't figured out yet is if SIMD vectorization applies equally as well to `Ref` as it does with `Block`.  I guess other code we've already written assumes that it does?

... and given that discussion, I'm not sure why _any_ code in the framework should prefer `Block` over `Ref` -- maybe it should be `Ref` everywhere?  @david-german-tri Any thoughts from the origin of `Block` way back when?

sherm1, 2017-02-06:
> I'm not sure why any code in the framework should prefer Block over Ref -- maybe it should be Ref everywhere?

I don't think Ref is suitable as a function return when a segment() or block() of a larger object is being returned since there would be no object available for it to reference. So I *think* we would want Blocks as returns and Refs and parameters. But I lack certainty!

jwnimmer-tri, 2017-02-06:
Both `Ref` and `Block` are aliases into pre-existing storage and are identical with respect to that concern.

sherm1, 2017-02-06:
Block has members `m_startRow`, `m_startCol`, and `m_outerStride` as well as the reference (`m_xpr`) to the underlying object.  Ref has `m_data`, `m_rows/cols`, and stride info. For a block, m_data has to point somewhere into the middle of the referenced object rather than to the object itself. In that case I *think* a subsequent attempt to make a sub-block would fail because the column spacing would be misapplied because m_data points to an arbitrary row.

That could be a reason to prefer returning a Block rather than a Ref -- it appears more flexible.

jwnimmer-tri, 2017-02-07:
... that seems to not apply for `VectorXd` though, which is all we use in the framework?

sherm1, 2017-02-07:
> that seems to not apply for VectorXd though

I think that's right. But ... I suppose the same argument applies to alignment. Block maintains the alignment since it refers to the original object. Ref's data starts at an arbitrary offset. But then maybe Ref is more efficient if alignment is not helping: Block's accessors presumably modify indices and then invoke the original object's accessors. Ref goes directly to the raw data. Hard to say which would be better given unknown impact of inlining and alignment.

My (mild) feeling is that `Block` is less weird than `Ref` and is a better type to return if you know you actually have a block.

david-german-tri, 2017-02-07:
> My (mild) feeling is that Block is less weird than Ref and is a better type to return if you know you actually have a block.

To the best of my recollection, this was my original thinking as well.  I'm not deeply committed to it, but would also rather not churn a bunch of code without a compelling reason.

jwnimmer-tri, 2017-02-18:
Update: I've test-ported enough code that I'm fairly confident in the APIs.  I've started work on steeling a first PR into shape at https://github.com/jwnimmer-tri/drake/tree/systems-single-base-pr1.

RussTedrake, 2017-05-27:
I believe this is resolved

closed 2017-05-27 by RussTedrake