RobotLocomotion/drake #4452: BulletModel::newBulletStaticMeshShape leaks

jwnimmer-tri, 2016-12-10:
The code, roughly:
```
std::unique_ptr<btCollisionShape> BulletModel::newBulletStaticMeshShape(...) {
...
  btTriangleMesh* mesh_interface = new btTriangleMesh();
...
  btBvhTriangleMeshShape* bvh_mesh_shape =
      new btBvhTriangleMeshShape(mesh_interface, useQuantizedAabbCompression);
  std::unique_ptr<btCollisionShape> bt_shape(bvh_mesh_shape);
```

This would appear to pass ownership of the `mesh_interface` memory to `bvh_mesh_shape`, but the `btBvhTriangleMeshShape` does not `delete` the `mesh_interface` upon description -- it is borrowing the pointer, not owning it.

Thus sayeth Bazel valgrind, roughly:
```
jwnimmer@call-cc:~/jwnimmer-tri/drake-distro$ bazel test --config memcheck //drake/multibody/collision:model_test
...
==27636== 39,268 (319 direct, 38,949 indirect) bytes in 1 blocks are definitely lost in loss record 7 of 9
==27636==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==27636==    by 0x4F3F18A: btAlignedAllocDefault(unsigned long, int) (btAlignedAllocator.cpp:70)
==27636==    by 0x420410: operator new (btTriangleIndexVertexArray.h:79)
==27636==    by 0x420410: DrakeCollision::BulletModel::newBulletStaticMeshShape(DrakeShapes::Mesh const&, bool) (bullet_model.cc:190)
==27636==    by 0x4210C3: DrakeCollision::BulletModel::addElement(DrakeCollision::Element const&) (bullet_model.cc:275)
==27636==    by 0x414522: DrakeCollision::(anonymous namespace)::ModelTest_StaticMeshes_Test::TestBody() (model_test.cc:825)
==27636==    by 0x4374E6: HandleSehExceptionsInMethodIfSupported<testing::Test, void> (gtest.cc:2078)
...
```

jwnimmer-tri, 2016-12-10:
#2710 is possibly related, where it mentions that leaking memory is a consequence of our collision library design deficiencies.

sherm1, 2016-12-10:
Thanks, @jwnimmer-tri. I believe this is exactly what @amcastro-tri was referring to in #2710 -- is that right, Alejandro? I reassigned both of these to @SeanCurtis-TRI.

amcastro-tri, 2016-12-13:
That is exactly right. Resolution of #2710 will take care of this and I know @SeanCurtis-TRI is already working on this.

jwnimmer-tri, 2016-12-16:
I guess in the meantime, I will add a suppression with a TODO citing this issue.

SeanCurtis-TRI, 2017-01-13:
What's the easy methodology for repeating your results (assuming I don't have bazel to "thus sayeth" to me?

jwnimmer-tri, 2017-01-13:
As a one-off it would be something like `valgrind build/somethingsomething/model_test`.  The full CMake instructions are http://drake.mit.edu/dynamic_analysis_tools.html#memcheck.

PR #4784 (2017-01-17): Patch memory leak for bullet triangle meshes
`BulletModel` owns all triangle meshes that it makes so that it is
responsible for cleaning them up.

(Incidentally renamed `mesh_interface` to `mesh` because it was more compatible with the declared types (e.g., `btTriangleMesh`).

fixes #4452

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4784)
closed 2017-01-17 by SeanCurtis-TRI