RobotLocomotion/drake #4464: Deal with ccache in PATH for bazel builds

jwnimmer-tri, 2016-12-12:
When a user has the magic `ccache` directory in their `$PATH`, bazel shows weird error messages that don't indicate the root cause of the problem and then fails.

Given that `ccache` is essentially required to have a good experience when using the CMake backends (ninja, make, etc.), and we can't (or at least shouldn't) have CMake inject ccache onto the path for us, we should do our best to accommodate developers who need to use both build systems from time to time.

If possible, we should either fail-fast in the case of `.../lib/ccache` on the path, or even better we should automatically remove the entry or otherwise fix the compiler variable(s) so that bazel is happy.

Part of #3129.

jwnimmer-tri, 2016-12-13:
FTR idea from another dev: if there's a good `ccache` env var to turn it off, that would be the easiest solution.  That would depend on the failure being at runtime not sandbox time, though.

jwnimmer-tri, 2016-12-14:
Related:
- bazelbuild/bazel#1164
- bazelbuild/bazel#1139
- bazelbuild/bazel#1322

I am trying a few things locally to discover the best motivation.

jwnimmer-tri, 2016-12-14:
Okay, while I think there may be `cc_toolchain` fixes to make this fail-fast or self-repair better, there doesn't seem to be a one-liner fix.  I'm going to FAQ it and return later.

jwnimmer-tri, 2016-12-20:
For now at least, I'll call the doc fix good enough.  I think a toolchain fix is likely to be better done in a PR to upstream code that in our project.

We can revisit if we find lots of Drake Developers complaining.

david-german-tri, 2017-02-10:
After a really epic onramp failure today, I'm reopening this.

david-german-tri, 2017-02-10:
https://github.com/bazelbuild/bazel/issues/1143 is an obstacle.

PR #5130 (2017-02-11): Make the Bazel build fail fast if ccache is on the path.
@ggould-tri for feature review, @jwnimmer-tri for platform review

Fixes #4464 

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5130)
closed 2017-02-11 by david-german-tri