RobotLocomotion/drake #4468: Mysterious CI Error: Could NOT extract MOSEK from archive

liangfok, 2016-12-12:
# The Error

```
[drake:make] CMake Error at CMakeLists.txt:127 (message):
05:50:57 
                                                                                
[drake:make]   Could NOT extract MOSEK from archive
05:50:57 
```

# Example Logs

* https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-clang-nightly-ros-release/139/

# Prognosis

Maybe the HTTP server hosting MOSEK went down? I also see the following lines from the console log, which indicate MOSEK is being downloaded as a tarball:

```
[drake:make] -- Downloading the MOSEK archive from http://download.mosek.com/stable/7/mosektoolslinux64x86.tar.bz2
05:50:57 
```

david-german-tri, 2017-04-13:
@stonier This is a tracking issue for a CI flake.  In my view, no action is required, and it will likely be moot after CMake is turned down.

jamiesnape, 2017-08-07:
Closing because CMake builds no longer use MOSEK.

closed 2017-08-07 by jamiesnape