RobotLocomotion/drake #4469: Warning - SWIG / Python Detected a Memory Leak

liangfok, 2016-12-12:
# Problem Definition

The following warning is being printed when the SWIG / Python `test_python_pydrake.test.testPR2IK` unit test is run:

```
swig/python detected a memory leak of type 'std::map< std::string,int,std::less< std::string >,std::allocator< std::pair< std::string const,int > > > *', no destructor found.
```

# Examples:
* https://drake-cdash.csail.mit.edu/testDetails.php?test=5480982&build=251903

jwnimmer-tri, 2016-12-13:
I'm not sure why we care?

liangfok, 2016-12-13:
We might not care depending on whether we decide to switch away from SWIG: https://github.com/RobotLocomotion/drake/issues/4486.

jwnimmer-tri, 2016-12-13:
I mean, say the pydrake bindings leak a little memory.  Do we even care about that?  This is just a console message, not a memcheck warning, not an anything.  I guess it's confusing in the build output, but beyond that I don't see why we care.

liangfok, 2016-12-13:
Are you saying this warning is not indicative of a defect in C++ Drake?

jwnimmer-tri, 2016-12-13:
Right.  It's specific to SWIG.

liangfok, 2016-12-13:
Oh great. That's good to know. I'll close this issue for now pending the resolution of #4486.

closed 2016-12-13 by liangfok

jamiesnape, 2016-12-13:
> not a memcheck warning,

Note that memcheck screams so much when testing the Python bindings that we just switched them off.

rdeits, 2016-12-13:
This is a symptom of swig being asked to return a type that it hasn't created a destructor for. It's easy to fix on a case-by-case basis, but only if we care enough about making swig work.

PR #4949 (2017-02-13): Replace SWIG bindings with Pybind11, and remove SWIG dependency
~~This isn't quite ready for review yet, but I want to see how it does on CI. This also includes the changes from #4944 which should come in first. I'll update this branch when that happens.~~

This replaces all of the `pydrake` bindings with identical bindings generated with Pybind11. With this change and #4944 we no longer have any dependencies on SWIG, so this also deletes the SWIG and swigmake externals. 

This also:

* changes the python binding build process, so there should no longer be any python build files in the source  [... cut]