RobotLocomotion/drake #4471: steering_command_driver.py Should Be Updated To Use Realistic Acceleration Profiles

liangfok, 2016-12-12:
#Problem Definition

`drake/automotive/steering_command_driver.py` currently uses hard-coded and linear acceleration / deceleration profiles when the user hits and holds the up and down keyboard arrows. The shape and slope of these acceleration / deceleration profiles is not realistic. This issue tracks the enhancement of this program to use realistic acceleration / deceleration profiles.

Another issue with `steering_command_driver.py` is the fact that the reference velocity outputted by `steering_command_driver.py` immediately becomes zero when the user releases the forward or backward key. This results in a "step function" in the reference velocity signal for the vehicle's controllers to follow, which is sub-optimal and may result in controller instability.

/CC: @shensquared - Just wanted you to see this; not assigning it to you since this is a low priority task.

jwnimmer-tri, 2016-12-12:
I mean, the throttle message is batshit anyway.  Sometimes it means target velocity, sometimes it means acceleration.  I feel like we should fix the system models before we fix the key/joy input tool.

liangfok, 2016-12-12:
Good point. I created #4472 and gave it a higher priority.

liangfok, 2016-12-12:
Looking at `SimpleCar`'s usage of `lcmt_driving_command_t`  (per @jwnimmer-tri's comment here: https://github.com/RobotLocomotion/drake/issues/4472#issuecomment-266495303), I now see that its `throttle` and `brake` values are actually gains applied to the maximum acceleration / deceleration of the vehicle. Thus, their units are "change in the acceleration per simulation cycle". Thus, the linear increase and decrease in the throttle / brake values that are currently implemented in `steering_command_driver.py` are actually specifying the change in the acceleration per update cycle, which I believe is called the "jerk".

jwnimmer-tri, 2017-03-17:
Now that we almost have a suite of demos working, we should revisit this in a few weeks.

liangfok, 2019-06-08:
Closing this since it's been superseded by other steering wheel drivers.

closed 2019-06-08 by liangfok