RobotLocomotion/drake #4472: lcmt_driving_command_t Should Be Updated to Include Reference Velocity Rather Than Reference Throttle / Brake

liangfok, 2016-12-12:
# Problem Definition

`drake/lcmtypes/lcmt_driving_command_t.lcm` over-specifies the system's desired state by including both `throttle` and `brake` reference values.

```
struct lcmt_driving_command_t {
  // The timestamp in milliseconds.
  int64_t timestamp;

  double steering_angle;
  double throttle;
  double brake;
}
```

In reality, the `car_sim_lcm` system only provides two control inputs: (1) desired steering angle and (2) desired velocity. We should thus modify `lcmt_driving_command_t` to specify the desired velocity rather than the desired state of the `throttle` and `brake`. Once this is done, `car_sim_lcm` can be simplified since it no longer needs to combine `throttle` and `brake` commands into a single reference `velocity` for the controller to follow.

jwnimmer-tri, 2016-12-12:
Careful.  The `SimpleCar` model uses this message as true throttle & brake; see https://github.com/RobotLocomotion/drake/blob/21c72d82e859e9367c5f1dd75b39673af93d1ff2/drake/automotive/simple_car.cc#L117.

IIRC @RussTedrake was going to rewrite the xdot algebra, but I can't quite recall if the input remains `a` or `v_desired`.

liangfok, 2016-12-12:
I believe the SimpleCar code can be simplified to just use a single "desired velocity" value. Here is the original code:

```
  // Apply simplistic throttle.
  T new_velocity =
      state.velocity() + (input.throttle() * config_.max_acceleration() *
                          config_.velocity_lookahead_time());
  new_velocity = min(new_velocity, config_.max_velocity());

  // Apply simplistic brake.
  new_velocity += input.brake() * -config_.max_acceleration() *
                  config_.velocity_lookahead_time();
  new_velocity = max(new_velocity, static_cast<T>(0.));
```

Is the following code identical?

```
  // Apply simplistic throttle.
  T new_velocity =
      state.velocity() + (input.throttle() - input.brake()) * config_.max_acceleration() *
      config_.velocity_lookahead_time());
  new_velocity = min(new_velocity, config_.max_velocity());
  new_velocity = max(new_velocity, static_cast<T>(0.));
```

_Disclaimer: The above may be completely wrong b/c I just quickly skimmed the code and didn't test anything._

liangfok, 2016-12-12:
Upon further inspection, `lcmt_driving_command_t`'s `brake` and `throttle` fields should simply be interpreted as gain for selecting the desired change in acceleration / deceleration of a vehicle. In other words, they are currently being interpreted as a "normalized jerk". In this case, perhaps we should replace the `brake` and `throttle` fields with a single `normalized_jerk` field. I've updated the name of this issue to reflect this new direction.

RussTedrake, 2016-12-12:
my version is here:
https://github.com/RussTedrake/drake/blob/simplecar/drake/automotive/simple_car.cc#L112

I could actually PR it today/tomorrow... now that the if_then_else work has landed.

liangfok, 2016-12-15:
Problem: `lcmt_driving_command_t` is an auto-generated file and the generator, `drake-distro/drake/tools/lcm_vector_gen.sh` does not support adding comments for each field in the message. We're going to need to modify the generator to support comments to resolve this issue.

PR #4579 (2016-12-22): Adds Documentation Comments to the Generated LCM Messages.
Resolves #4472.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4579)
closed 2016-12-22 by jwnimmer-tri