RobotLocomotion/drake #4478: Octomap library linking problem

ljarin, 2016-12-12:
I'm trying to build DRAKE using ROS indigo on ubuntu 14.04. The build fails in the drake package due to some sort of linking problem with an octomap library. Here is the most relevant portion of the terminal output:

```
CMakeFiles/octomapWrapper.dir/octomapWrapper.cpp.o: In function `octomap::OccupancyOcTreeBase<octomap::OcTreeNode>::updateNodeRecurs(octomap::OcTreeNode*, bool, octomap::OcTreeKey const&, unsigned int, float const&, bool)':
/opt/ros/indigo/include/octomap/OccupancyOcTreeBase.hxx:386: undefined reference to `octomap::OcTreeNode::createChild(unsigned int)'
CMakeFiles/octomapWrapper.dir/octomapWrapper.cpp.o: In function `octomap::OccupancyOcTreeBase<octomap::OcTreeNode>::readBinaryNode(std::istream&, octomap::OcTreeNode*) const':
/opt/ros/indigo/include/octomap/OccupancyOcTreeBase.hxx:991: undefined reference to `octomap::OcTreeNode::createChild(unsigned int)'
/opt/ros/indigo/include/octomap/OccupancyOcTreeBase.hxx:1008: undefined reference to `octomap::OcTreeNode::createChild(unsigned int)'
/opt/ros/indigo/include/octomap/OccupancyOcTreeBase.hxx:1003: undefined reference to `octomap::OcTreeNode::createChild(unsigned int)'
/opt/ros/indigo/include/octomap/OccupancyOcTreeBase.hxx:986: undefined reference to `octomap::OcTreeNode::createChild(unsigned int)'
CMakeFiles/octomapWrapper.dir/octomapWrapper.cpp.o:/opt/ros/indigo/include/octomap/OccupancyOcTreeBase.hxx:996: more undefined references to `octomap::OcTreeNode::createChild(unsigned int)' follow
collect2: error: ld returned 1 exit status
make[5]: *** [/home/la27059/drake/src/drake/drake/matlab/systems/robotInterfaces/octomapWrapper.mexa64] Error 1
make[4]: *** [matlab/systems/robotInterfaces/CMakeFiles/octomapWrapper.dir/all] Error 2
```

There were some other warnings that did not seem too related. The files in my /opt/ros/indigo/include/octomap look to be alright, and I have used octomap in my own ros packages before. I also tried uninstalling and reinstalling ros-indigo-octomap-ros. Let me know if there is a better place to ask this question - I suspect this is something I am doing wrong, as opposed to an "issue."

jwnimmer-tri, 2016-12-12:
Possibly a duplicate of #3814?

ljarin, 2016-12-12:
Oops, yeah looks like it.

jwnimmer-tri, 2016-12-12:
No problem.  It looks like that discussion has much more content, so I'll close this one.

closed 2016-12-12 by jwnimmer-tri