RobotLocomotion/drake #4485: Investigate Bazel support for CLion

jwnimmer-tri, 2016-12-13:
Many Drake developers use CLion.  We should figure out what works and doesn't with Bazel BUILD * WORKSPACE files, consider what improvements we can make, and document the results.

Part of #3129.

jwnimmer-tri, 2016-12-15:
Possibly relevant bazelbuild/bazel#1029 which pointed me to bazelbuild/intellij#26 :confetti_ball:.

RussTedrake, 2016-12-17:
fwiw, just lost a few hours trying Tulsi with bazel to work with Xcode.  Long story short, I think it's not going to support our workflow without some work.  I made some local edits to the tulsi code to get fairly far along (https://github.com/RussTedrake/tulsi), but it is not able to understand the "@glib://" deps, which apparently comes in through pkg-config (which comes in through the thirdparty bazel support).  

The local edits that I did make were needed for it to even consider cc_* as valid targets.  It was only looking for e.g. ios_application.  So it's clearly not being used for our workflow yet.

jwnimmer-tri, 2017-01-12:
IIRC possibly David was going to try to look into this also.

david-german-tri, 2017-01-13:
Yes, I'm trying to get https://github.com/bazelbuild/intellij/issues/26 working locally.  No luck yet.

david-german-tri, 2017-01-19:
Fixed in #4814.

closed 2017-01-19 by david-german-tri