RobotLocomotion/drake #4486: Reconsider the use of SWIG

liangfok, 2016-12-13:
# Problem Definition

Numerous oddities involving SWIG have cropped up:

* #4469 
* #4454 
* #4243 
* #2646

# Action Item

We should reconsider whether to continue using SWIG or switch to something else that is perhaps more target language specific. For example, for Python bindings, @jwnimmer-tri recommended [Boost::Python](http://www.boost.org/doc/libs/1_62_0/libs/python/doc/html/index.html) and @jamiesnape recommended [pybind11](https://github.com/pybind/pybind11).

jamiesnape, 2016-12-13:
If we use Boost for anything else, we should use Boost::Python without hesitation. The two options above are very, very similar otherwise, so I have no real preference.

We have the long-standing Boost and MATLAB issues to consider, of course.

rdeits, 2016-12-13:
The additional issue is whether you want the Matlab bindings to be identical to the Python bindings and how much work you want to do to make that happen.

jamiesnape, 2016-12-16:
@mwoehlke-kitware is going to investigate this and post a proposal here for @david-german-tri, @rdeits, and others to review.

Initial thoughts are that MATLAB bindings remaining SWIG are fine, and whether we use Boost::Python depends on whether MATLAB needs to be simultaneously enabled with the Python bindings, and whether we bring in Boost for other reasons, e.g., OMPL.

david-german-tri, 2017-01-05:
Bumping to high priority because it's blocking #2482.

From f2f discussion: We believe that `Boost::Python` only needs to be linked into the Python bindings at the leaf of the build, and thus won't wind up linked into mex files.  If that turns out to be incorrect, we can require that users build MATLAB or Python bindings but not both simultaneously.  Thus, this is unblocked.

rdeits, 2017-01-05:
@david-german-tri can you explain how it's blocking #2482? It's not obvious to me, since SWIG is unrelated to Pods.

david-german-tri, 2017-01-05:
@jamiesnape's belief is that the remaining few lines of https://github.com/RobotLocomotion/drake/blob/master/cmake/pods.cmake, which fiddle with `RPATH`, are necessary only to support the current SWIG approach.  (I haven't actually tested that.)

rdeits, 2017-01-05:
I see. But surely SWIG can be used just fine without pods, since that's widely done in practice.

jamiesnape, 2017-01-05:
There are other issues. That is just the most pressing.

jwnimmer-tri, 2017-01-12:
@mwoehlke-kitware -- #4771 has a question about the roadmap here.  I believe there is additional information and decisions made but not yet scribed to the issue.  Could you post a write-up?

jamiesnape, 2017-01-13:
We are going to use Boost::Python and the port will begin as soon as Director tests are up and running on CI, probably next week.

mwoehlke-kitware, 2017-01-16:
TBH, I wish Shiboken wasn't so in flux at the moment; it's the only binding generator I can find that doesn't require duplicating the entire API you want to expose.

On further investigation, between Boost::Python and pybind11, I'd actually rather use pybind11. The API is more modern and much less macro-heavy, the documentation seems easier to understand, how to integrate it into the build is much more obvious and better documented (Boost::Python basically says "use bjam or it will break"), and it just generally feels less brittle and better designed.

jamiesnape, 2017-01-16:
We have to introduce Boost for #3413, so Boost::Python would be preferable than added the additional dependency unless we really have to.

jwnimmer-tri, 2017-01-16:
TRI has Bazel build support for `boost::python` that we can probably furnish to be open-sourced, if that's helpful.  The code has a ` WORKSPACE` entry for the Boost targz source release, then a small `BUILD` file that compiles a `cc_library` for `Boost::Python`, and then you just write your own cc file with the `BOOST_PYTHON_MODULE` call for the module you're going to export.

closed 2017-02-13 by rdeits

EricCousineau-TRI, 2017-05-03:
Linking to #1267.

PR #4949 (2017-02-13): Replace SWIG bindings with Pybind11, and remove SWIG dependency
~~This isn't quite ready for review yet, but I want to see how it does on CI. This also includes the changes from #4944 which should come in first. I'll update this branch when that happens.~~

This replaces all of the `pydrake` bindings with identical bindings generated with Pybind11. With this change and #4944 we no longer have any dependencies on SWIG, so this also deletes the SWIG and swigmake externals. 

This also:

* changes the python binding build process, so there should no longer be any python build files in the source  [... cut]