RobotLocomotion/drake #4487: Need a DiscretePidController

liangfok, 2016-12-13:
# Problem Definition

Per F2F discussion with @sammy-tri and @naveenoid, the existing `drake::systems::PidController` (implemented in `drake-distro/drake/systems/controllers/pid_controller.h`) runs at the same update frequency as the plant that it is controlling. This is problematic because it prevents us from increasing the damping (i.e., derivative / Kp) gains by any meaningful amount without the plant going unstable. Without being able to increase the damping gains, we cannot fine-tune the PID controller gains to match the actual hardware.

# Proposed Solution

Fundamentally, we want to cycle the `PidController` at a frequency that is less than the frequency at which we update the `RigidBodyPlant`. This better reflects reality since in reality the plant is being updated at a frequency of infinity. One way to do this would be to have a separate `ControllerExectutor` that runs separately from the existing integrator as described in #4342. However, another potentially easier method is to make a `DiscretePidController` that is configured to only issue new torque commands upon discrete updates (i.e., within method `DoEvalDiscreteVariableUpdates()`). For an example, see how `drake-distro/drake/examples/schunk_wsg/schunk_wsg_lcm.cc` calls `this->DeclareUpdatePeriodSec(0.05);` in its constructor, and its `DoEvalDiscreteVariableUpdates()` method.

david-german-tri, 2016-12-13:
FWIW I think `DiscretePidController` is both much less work and more conceptually sound, but I believe either approach is viable from a System2 standpoint.

liangfok, 2016-12-14:
@david-german-tri: Do you think we should _replace_ the existing PidController with one that is discrete?

liangfok, 2016-12-14:
@sammy-tri at what discrete frequency were you running the controller?

sammy-tri, 2016-12-14:
200hz, 100hz, 50hz.  Haven't spent much time on it yet.

david-german-tri, 2016-12-14:
> Do you think we should replace the existing PidController with one that is discrete?

Continuous PID controllers are a real thing people like to build, so I think we should continue to have a model of them, no?  (And if the model is wrong, it should be fixed as and when there's a need.)

liangfok, 2016-12-14:
During a F2F meeting with @RussTedrake, we've decided to wrap the current `PidController` in a zero-order-hold system to mimic a controller that's updating slower than the plant. The reason we want to keep the "continuous" version of the `PidController` is because it will allow more powerful analysis to be performed on the overall system. Because of this, I shall close this issue.

closed 2016-12-14 by liangfok

RussTedrake, 2016-12-14:
just to be clear, I commented that 
1) we are likely to make some generic tools for creating sampled-data approximation of systems (e.g. wire up zero-order-hold outputs, etc.  and the PID doesn't feel like a special enough case to go around the more general approach.
2) the continuous time abstraction is much simpler to reason about, and that we don't want to add the modeling complexity of the sample-data system until we need to (typically when the sampling rate starts to get close to the fundamental time constants of the closed-loop system).