RobotLocomotion/drake #4488: Move WorldSimTreebuilder into General Location

liangfok, 2016-12-13:
During a F2F meeting with @sammy-tri and @naveenoid, we decided that the C++ class `drake::examples::kuka_iiwa_arm::WorldSimTreeBuilder` is of general use and should thus be moved into a more general location. Functionally, it is similar to `AddFlatTerrainToWorld()`, which is defined in `drake-distro/drake/multibody/rigid_body_tree_construction.h`. One option would be to move `WorldSimTreeBuilder` to also be in `drake-distro/drake/multibody/rigid_body_tree_construction.h`, but I'm generally opposed to storing a class in a file that's not named after the class (since it sometimes makes the class more difficult to find). Thus, I propose creating a new directory called `drake-distro/drake/multibody/construction/` that contains two files:

1. add_flat_terrain_to_world.h
2. world_sim_tree_builder.h

Thoughts?

jwnimmer-tri, 2016-12-13:
Generalize at the rule of three (or possibly two with a good argument for what the third is).  What other two portions of Drake want this utility?

liangfok, 2016-12-13:
It's currently being used by @naveenoid's KUKA world demo, and @sammy-tri's KUKA + Gripper demo. I would like to use in the HSRb demos soon.

liangfok, 2016-12-13:
It will likely be used by the automotive demos once we start simulating more complex environments.

jwnimmer-tri, 2016-12-13:
So it sounds like "KUKA" and "HSRb" are the two uses, so that's a good start.  But I'm surprised the HSRb stuff wants programmatic building of scenes -- I thought it was mostly ROS, which means using an SDF to specify the world.

liangfok, 2016-12-13:
For the HSRb, I'd like to cherry-pick the best of both worlds (Drake vs. ROS). For example, I'm hoping we can support programmatic world creation, then provide a method like `std::vector<std::string> RigidBodyTree::GenerateUrdfs()`, which returns a set of URDFs representing the world. These URDFs can then be loaded onto the ROS parameter server for use by all of the other sibling ROS nodes.

FTR, the reason I mention URDF instead of SDF above is because most of ROS is still using URDF. It's Gazebo that has officially migrated to SDF.

sammy-tri, 2017-12-11:
This was effectively fixed by #6864

closed 2017-12-11 by sammy-tri