RobotLocomotion/drake #4500: Generalize RigidBodyPlant's Input and Output Ports To be Model Instance Specific

liangfok, 2016-12-14:
# Problem Definition

Currently, the `RigidBodyPlant` has one input port containing the effort command for _all_ actuators in the plant's `RigidBodyTree`, and one output port containing the _entire tree's_ configuration. This design is cumbersome when there are numerous model instances within the `RigidBodyTree` and each model instance needs to be controlled by a different controller. Using the current design, we will need to apply multiplexers to create the input to the `RigidBodyPlant` and de-multiplexers to split the output of the `RigidBodyPlant`. Such a design is being employed in #4475.

# Proposed Solution

Consider modifying `RigidBodyPlant` to provide `N` input ports and `N` output ports, where `N` is the number of model instances within the tree. Each input port and output port pair contains the effort command and current configuration of _a particular model instance_. By having `N` pairs of input / output ports, we can connect `N` controllers, each focusing on controlling one model instance within the `RigidBodyTree`.

For backwards compatibility, we can provide an extra input / output port pair that is the same as the current input / output ports of `RigidBodyPlant` (i.e., they take the full effort command vector for all actuators in the `RigidBodyTree` and output the `RigidbodyTree`'s full generalized state vector).

SeanCurtis-TRI, 2016-12-14:
Philosophically, this makes sense.  Practically, is this arising now because we're abusing the rigid body plant to move non-physics cars?  (And feel free to ignore this if I'm way off base.)

sherm1, 2016-12-14:
This sounds very reasonable but needs @amcastro-tri to comment -- delay likely due to conference attendance.

liangfok, 2016-12-14:
> Practically, is this arising now because we're abusing the rigid body plant to move non-physics cars?

In the above-linked #4475, it's because we're using two separate controllers on a single `RigidBodyPlant`: one for the IIWA and one for the gripper. Both are physics-based.

@RussTedrake has a WIP project where numerous `SimpleCar` vehicles are loaded into a `RigidBodyTree` and each vehicle needs to be controlled separately. This one uses non-physics cars.

From the above, the need for model-instance-specific input/output ports appears to span physics and non-physics-based simulations.

SeanCurtis-TRI, 2016-12-14:
Groovy.  I have no concerns.

sammy-tri, 2016-12-14:
Candidate proposal in #4506

amcastro-tri, 2016-12-14:
This sounds reasonable to me. #4506 also makes sense to me. Still reviewing.

PR #4506 (2016-12-20): Give RigidBodyPlant per-system input and output ports
Fixes #4500 

This is a draft implementation for preliminary review for those interested.  I'm not 100% sure it's merge-worth as it is right now.

I have tested it by porting my iiwa+WSG simulation to this interface and it appears to work identically to #4475 (as I'd expect).

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4506)
closed 2016-12-20 by sammy-tri