RobotLocomotion/drake #4502: LcmPublisherSystem should only publish at a fixed rate

david-german-tri, 2016-12-14:
This is now possible in System2/Simulator2 using `CalcNextUpdateTime`.

david-german-tri, 2016-12-14:
Ditto for DrakeVisualizer.

RussTedrake, 2016-12-16:
I agree it should be an option, but I’m not sure that this is obviously the right default behavior.  Use the default publish (e.g. any major timestep of the simulator) has the advantage that we publish anytime the simulation decides something.  It’s the best way to keep a downstream lcm subscriber “in step” with the simulation… no?

david-german-tri, 2016-12-16:
@RussTedrake Sounds good to me.  I have no particular intuition about which is better.  As a bonus, your proposal is a little easier to implement, since it won't require custom [do_publish](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/framework/system.h#L40) callbacks.

jwnimmer-tri, 2016-12-22:
So #4111 seems to cover (duplicate) the Drake Visualizer System portion of this.

PR #4731 (2017-01-12): Modifies LcmPublisherSystem to optionally throttle its publish rate.
Resolves #4502.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4731)
closed 2017-01-12 by liangfok