RobotLocomotion/drake #4504: Building against deleted header files

SeanCurtis-TRI, 2016-12-14:
Take this issue modulo the actual answer.

We've encountered an inconvenient build artifact.  The problem is as follows:

  - Drake is built, populating `install` directory.
  - Upstream changes remove a referenced header file from repository.
  - I update my local repo to match the upstream repo (the header disappears from source branch).
  - I can still build because, apparently, the `install` path is part of the include path.
  - `make clean` does not flush the `install` directory -- thus, we still have code that conflicts in reality, but isn't detected in the local build.

Possible solutions:
   1) Build system shouldn't be looking in the `install` directory (although, I can imagine for we're going there for generated headers???)
   2) Have `make clean` flush `install` directory.

liangfok, 2016-12-14:
Does this problem occur using both CLion and command line?

jwnimmer-tri, 2016-12-14:
FTR, I have been bitten by this too.  I have chosen to use Bazel to get correctness, rather than fix our use of CMake.

SeanCurtis-TRI, 2016-12-14:
@liangfok Yep.  Definitely CLion -- can't definitively state command line, but I suspect so.  I can't imagine that CLion would unilaterally extend its include paths.  (And as @jwnimmer-tri has been bitten...it seems to be a very democratic issue; everyone gets some, regardless of IDE.)

RussTedrake, 2016-12-16:
I can confirm that this has been a known issue for a long, long time.

david-german-tri, 2017-04-03:
#3129 makes this moot, so closing.

closed 2017-04-03 by david-german-tri