RobotLocomotion/drake #4505: Can't apply RK3 integrator to iiwa_world_demo

SeanCurtis-TRI, 2016-12-14:
Attempted to swap integrators in the `iiwa_world_demo` to RK3 and failed.

Modifications:

Original code
```
auto simulator = std::make_unique<systems::Simulator<double>>(*demo_plant);
demo_plant->SetDefaultState(simulator->get_mutable_context());
simulator->Initialize();
simulator->StepTo(FLAGS_simulation_sec);
```

New code
```
auto simulator = std::make_unique<systems::Simulator<double>>(*demo_plant);
demo_plant->SetDefaultState(simulator->get_mutable_context());

// change integrator
simulator->reset_integrator<RungeKutta3Integrator<double>>(*demo_plant,
                                                             simulator->get_mutable_context());
simulator->get_mutable_integrator()->request_initial_step_size_target(1e-3);

simulator->Initialize();
simulator->StepTo(FLAGS_simulation_sec);
```

Error message:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failure at /home/seanc/code/drake-distro/drake/../drake/systems/framework/system.h:648 in DoMapQDotToVelocity(): condition 'generalized_velocity->size() == n' failed.
```

sherm1, 2016-12-14:
@edrumwri, this model has a robot arm (all pin joints with qdot=v) but also has some free bodies that have quaternions. Presumably something didn't override the map qdot to velocity method.

sherm1, 2016-12-14:
@amcastro-tri, I think this means RBPlant didn't override properly?

edrumwri, 2016-12-18:
Thanks for providing such a clear, fast way for me to reproduce the bug, @SeanCurtis-TRI.

I verified that there is indeed a subclass in there that isn't overriding properly (I was worried that this issue might be due to PR #4532). PassiveVisualizedPlant derives from Diagram, and it appears that one of the two (either the diagram abstract class or PassiveVisualizedPlant) *should* be overriding the map qdot to velocity method and is not. 

The reason that the problem is just now manifesting is that the default integrator does not need to call that method (the RK3 integrator uses the method for error norm estimation). RigidBodyPlant has been verified to produce the proper overriding behavior in runge_kutta3_integrator_test.cc.

Thoughts on this @sherm1? I have no experience with Diagram<T>, so further insight might help me to resolve this.

RussTedrake, 2016-12-18:
The problem is that Diagram implements DoMapVelocityToQDot but does not yet implement DoMapQDotToVelocity.  My guess is that it would be trivial to implement because at the diagram level it should be identical to the existing DoMapVelocityToQDot .  Cc @david-german-tri .

edrumwri, 2016-12-18:
Thanks for the hint, Russ- that indeed does not look challenging to
implement and test. I'll work on this unless @david-german-tri really wants
to.

david-german-tri, 2016-12-19:
> My guess is that it would be trivial to implement because at the diagram level it should be identical to the existing DoMapVelocityToQDot 

Sounds right to me.

> I'll work on this unless @david-german-tri really wants to.

It's all yours!

edrumwri, 2016-12-25:
Fixed in PRs #4536 and #4537.

closed 2016-12-25 by edrumwri