RobotLocomotion/drake #4508: Clean up MathematicalProgram

hongkai-dai, 2016-12-15:
I like the current structure of `MathematicalProgram` object, that it is flat without any derived class. Also I think it makes a lot of sense to have a separate `MathematicalProgramSolverInterface`, that we can explicitly use a specific solver (SNOPT, Mosek, Gurobi, etc) to solve a problem. I like to keep this structure untouched.

I imagine the user can solve an optimization problem in two ways

1. If the user does not want to think about which solver to use, then he/she can call `MathematicalProgram::Solve()` function, and `MathematicalProgram` would designate a solver.
2. If the user does want to use a specific solver, then he/she can call
   ```
   MathematicalProgram prog;
   SnoptSolver snopt_solver;
   snopt_solver.solve(prog);
   ```

With the imagined user scenario, I propose we make the following changes to `MathematicalProgram`.

1. [ ] 1. Each solver can return the solution in its own struct. For example, each solver can contain the exit code of that solver (INFO 41 in SNOPT, 10006 in Mosek, etc), and some solver can return the active constraints, dual variables, etc. To do so, we can add a new `MathematicalProgramSolverResult` class as the base class, with derived class `SnoptResult`, `MosekResult`, etc. `MathematicalProgramSolverInterface` has a abstract function
   ```C++
   const MathematicalProgramSolverResult& MathematicalProgramSolverInterface::solve(const MathematicalProgram& prog);
   ```
   The `Solve` function in `MathematicalProgram` is unchanged, that the returned `SolutionResult` only says whether the problem converges or not, without any extra information that is solver-specific.
2. [ ] 2. Currently we can call `SetSolverOption` or `GetSolverOptions` in `MathematicalProgram`. I think it has the following problems
   1. These public functions contradict the assumption that the user doesn't care about which solver to use, if he/she calls `MathematicalProgram::Solve()` function. 
   2. Moreover, the current signature for `SetSolverOption` is
      ```C++
      void MathematicalProgram::SetSolverOption(const std::string& solver_name, const std::string& solver_option, double option_value);
      ```
      This supposes that the `solver_option` is a string, but some solver does not use a string for its option. For example, the type for Mosek option is **enum**.
   My proposal is to move this `SetSolverOption` function to each solver, like `SnoptSolver`, `MosekSolver`, and they can use the `int` for the Mosek's solver option, while using `string` for Gurobi's solver option.
3. [ ] 3. The `MathematicalProgram.h` file is really big, almost 1600 lines. It would be good to downsize this file. I think we can move `Binding` outside of `MathematicalProgram` class, into its own file.

hongkai-dai, 2016-12-15:
@sammy-tri and @RussTedrake, would love to hear your opinion.

sammy-tri, 2016-12-15:
I like most of the above, but I do have one thought, and that's that there could be cases where multiple solvers could be used, and the calling code knows things about more than one.  Specifically, consider the IPOPT vs. SNOPT case, where IPOPT is always available and SNOPT requires a license.

In DirectTrajectoryOptimization::SolveTraj(), we set IPOPT options even though we'd prefer SNOPT as a solver if it's available.  I've been meaning to do something similar for IK planning, where IPOPT takes a long time to solve a problem with the defaults, but tuning some options would speed it up significantly (but we'd still prefer SNOPT, which the IK code has already tuned to solve problems quickly).

We could make these classes test which solvers are available and pick one at that level instead of letting MathematicalProgram:Solve() decide.  Though that gets even more complex if you have a (theoretical) case where an optional constraint requires a non-linear solver, but not all invocations of some planning task will have that constraint.

Sorry for not having a fully baked idea there, but I wanted to get the thought out there.  Will consider further.

hongkai-dai, 2016-12-15:
@sammy-tri that is a good point. I will change the second bullet point, that we still have `SetSolverOption` in `MathematicalProgram`, but the type for `solver_option` is templated, so that we can use either string or enum.

hongkai-dai, 2016-12-16:
Creating `MathematicalProgramSolverResult` does not work. I tried the following steps:

1. Create a Base class `MathematicalProgramSolverResult`.
2. Derive a new class `MosekSolverResult` from `MathematicProgramSolverResult`.
```
class MathematicalProgramSolverResult {
};
class MosekSolverResult : public MathematicalProgramSolverResult {
}
```

The problem is that I want `MathematicalProgramSolverInterface::Solve()` to return an instance of `MathematicalProgramSolverResult`. But this `Solve()` function is virtual, that should be overrided by the derived class, such as `MosekSolver`.  The following options do not work

1. Return a raw pointer
   ```C++
   class MathematicalProgramSolverInterface {
   ...
   virtual MathematicalProgramSolverResult* Solve(MathematicalProgram& prog) = 0// Dynamically allocate the result inside the function, returns the newly allocated pointer.
   }

   class MosekSolver : public MathematicalProgramSolverInterface {
   ... 
   MosekSolverResult* Solve(MathematicalProgram& prog)
   }
   ```
   It is a bad idea to manage the dynamically allocated raw pointer.
2. Return a unique_ptr
   ```C++
   class MathematicalProgramSolverInterface {
   ...
   virtual std::unique_ptr<MathematicalProgramSolverResult> Solve(MathematicalProgram& prog) // Dynamically allocate the result inside the function, returns the newly allocated pointer.
   }

   class MosekSolver : public MathematicalProgramSolverInterface {
   ... 
   std::unique_ptr<MosekSolverResult> Solve(MathematicalProgram& prog)
   }
   ```
   The compiler does not allow me to do this, since `std::unique_ptr<Derived>` cannot override `std::unique_ptr<Base>`.

sherm1, 2016-12-16:
I can show you how to get the unique_ptr option to work as you wrote it, Hongkai -- ask me tomorrow. But it will still be allocating on the heap.

hongkai-dai, 2018-12-21:
close the issue as it is overtaken by #9633.

For the length issue. A lot of them are repeated documentations. I think I can shorten the header file, by using named group in doxygen to remove redundant documentation.

closed 2018-12-21 by hongkai-dai