RobotLocomotion/drake #4519: MathematicalProgramSolverInterface wastes time parsing unchanged constraints, for model predictive control.

hongkai-dai, 2016-12-16:
<h1> Current implementation </h1>
Currently there are two main components for an optimization problem, a class `MathematicalProgram`, which stores the constraints/costs (`LinearConstraint`, `QuadraticConstraint`, etc), and a virtual class `MathematicalProgramSolverInterface`, with the derived class (`SnoptSolver, IpoptSolver, MosekSolver`, etc) for each solver.

The usage is like
```C++
MathematicalProgram prog;       // Empty optimization program.
prog.AddDecisionVariables(...); // Add decision variables.
prog.AddConstraint(...);        // Add constraints.
prog.AddCost(...);              // Add costs.
SnoptSolver snopt_solver;       // Create a solver
snopt_solver.Solve(prog);       // This function parses each constraints and costs, into the format that the specific solver can recognize.
```
 
We designed this separation between the `MathematicalProgram` and `MathematicalProgramSolverInterface`, so that the same program can be solved by different solvers (`SNOPT`, `IPOPT`, etc), and all these solvers can call the same shared `MathematicalProgram` object. For example.
```C++
SnoptSolver snopt_solver;
snopt_solver.Solve(prog);
IpoptSolver ipopt_solver;
ipopt_solver.Solve(prog);
```

<h1> Challenge </h1>

This approach works great, until we want to modify the optimization problem a little bit (change constraints/ costs), and then resolve the program, such as in the model predictive control. When calling `Solve` function in `MathematicalProgramSolverInterface` object, such as `MosekSolver.Solve()`, it builds up the constraints/costs in the Mosek format (`Mosek task` and `Mosek environment`), but these Mosek formats are not stored inside `MosekSolver`. So even if we just change one constraint in `MathematicalProgram`, we need to re-parse **ALL** the constraints if we call `MosekSolver::Solve()`. This can be a big waste of time, if we want to do model predictive control.

<h1> Ideas </h1>

`MathematicalProgram` should store the problem after being parsed by the solver, like `Mosek task` for Mosek, or `Gurobi model` for Gurobi. For each `Constraint` or `Cost` in `MathematicalProgram`, there should be a flag saying if it has been changed since last time the solver parses the constraint. So I propose make the following changes to `Binding` and `MathematicalProgram`
```C++
class Binding {
...
private:
  map<std::string solver_name, bool> dirty; // If dirty[solver_name] is true, it means this constraint needs to be parsed again by the solver, when calling the Solve() function.
}

class MathematicalProgram {
   ...
  private:
    MSKtask_t mosek_task_;    // Stores the parsed data for Mosek, gets updated after calling MosekSolver.Solve(MathematicalProgram& prog);
    GRBmodel gurobi_model_;   // Stores the parsed data for Gurobi, gets updated after calling GurobiSolver.Solve(MathematicalProgram& prog);
}
```

jwnimmer-tri, 2016-12-16:
How much of a performance hit is this (what fraction of time is spent reparsing vs solving)?  What do the profiler results say?

sammy-tri, 2016-12-16:
Is this (to some extent) what the largely unutilized MathematicalProgram::GetSolverData() was intended for?

hongkai-dai, 2019-03-26:
Close it for now, will reopen it if needed.

closed 2019-03-26 by hongkai-dai