RobotLocomotion/drake #4525: Mysterious LCM Error: Error copying file

liangfok, 2016-12-16:
# The Error

```
Error copying file "/Users/elcapitan/workspace/mac-clang-nightly-matlab-open-source-release/externals/lcm/lcm-java/jchart2d-code/ext/jide-oss-2.9.7.jar" to "/Users/elcapitan/workspace/mac-clang-nightly-matlab-open-source-release/build/externals/lcm/lcm-java/jchart2d-code/ext/jide-oss-2.9.7.jar".
make[5]: *** [lcm-java/jchart2d-code/ext/jide-oss-2.9.7.jar] Error 1
make[5]: *** Deleting file `lcm-java/jchart2d-code/ext/jide-oss-2.9.7.jar'
make[4]: *** [lcm-java/jchart2d-code/CMakeFiles/lcm-ext-jide-oss-jar.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
Error copying file "/Users/elcapitan/workspace/mac-clang-nightly-matlab-open-source-release/externals/lcm/lcm-java/jchart2d-code/ext/jide-oss-2.9.7.jar" to "/Users/elcapitan/workspace/mac-clang-nightly-matlab-open-source-release/build/externals/lcm/lcm-java/jchart2d-code/ext/jide-oss-2.9.7.jar".
make[5]: *** [lcm-java/jchart2d-code/ext/jide-oss-2.9.7.jar] Error 1
make[4]: *** [lcm-java/jchart2d-code/CMakeFiles/jchart2d.dir/all] Error 2
make[3]: *** [all] Error 2
```

# Example Logs

* https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-clang-nightly-matlab-open-source-release/227/

liangfok, 2016-12-16:
This appears to be a transient bug. I shall thus close it.

jonathan-owens, 2017-01-04:
It's back. See, e.g.: https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=291390

jwnimmer-tri, 2017-01-04:
Definitely not Liang's reponsibility, though.

liangfok, 2017-03-22:
https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-experimental-open-source/2022/

david-german-tri, 2017-04-13:
@stonier This is a tracking issue for a CI flake.  In my view, no action is required, and it will likely be moot after CMake is turned down.

PR #6738 (2017-09-29): Purge CMake not used by MATLAB
Toward #3129. Closes #3342, closes #3426, closes #4113, closes #4525, closes #5356, and closes #5572.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6738)
closed 2017-09-29 by jamiesnape