RobotLocomotion/drake #4526: Upgrade Bazel in CI to 0.4.2

jwnimmer-tri, 2016-12-16:
Upgrade Bazel in CI to 0.4.2 to match http://drake.mit.edu/developers.html#supported-configurations.  Right now we appear to have e.g. `  BAZEL_VERSION                  = 0.4.0-jdk7`.

Also required for #4518 to pass CI.

david-german-tri, 2016-12-16:
@jamiesnape @BetsyMcPhail any chance you could knock this out quickly today?  I think it's a prerequisite for turning on Bazel builds in pre-merge, which I'd also like to see happen today if possible.

BetsyMcPhail, 2016-12-16:
Logging in to get this done now - should be a quick update.

BetsyMcPhail, 2016-12-16:
The trusy and xenial AMIs have been updated to Bazel 0.4.2. 

The unprovisioned xenial script already installs Bazel 0.4.2.

See #4497 for Mac Bazel builds.

BetsyMcPhail, 2016-12-16:
I installed Bazel using JDK8 on Trusty and this is making the builds not happy (e.g. https://drake-jenkins.csail.mit.edu/view/Bazel/job/linux-gcc-bazel-experimental/56/console). It looks like a JDK7 version is available (although [deprecated]( https://bazel.build/versions/master/docs/install.html#ubuntu)). 

Unless I hear any objections I am going to go back and install the JDK7 version on the Trusy AMI.

jamiesnape, 2016-12-16:
It was JDK 7 before. The JDK 8 version will break Jenkins.

BetsyMcPhail, 2016-12-16:
Waiting for the Bazel experimental build but the trusty AMI should be good now.

david-german-tri, 2016-12-16:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-bazel-experimental/61/console

> FAILURE BECAUSE COULD NOT DETERMINE BAZEL VERSION

BetsyMcPhail, 2016-12-16:
Looking into it now

jamiesnape, 2016-12-16:
Both 0.4.1 and 0.4.2 are exiting with BLAZE_INTERNAL_ERROR even when calling `bazel help` and `bazel version`. Something is pretty broken somewhere.

jamiesnape, 2016-12-16:
We are having the same issue on local machines too.

BetsyMcPhail, 2016-12-16:
Changed back to the previous Trusty AMI (Bazel 0.4.0) in Jenkins for now.

jwnimmer-tri, 2016-12-16:
TTBOMK @david-german-tri is testing on Trusty on his workstation, so I'm not sure what's going on here.

jamiesnape, 2016-12-16:
Using JDK 7?

david-german-tri, 2016-12-16:
No, using JDK 8.

david-german-tri, 2016-12-16:
I'm pretty willing to drop official support for Bazel on Trusty if that gets us out of this fix, but we should take public comments on that.  I'll poll on slack.

david-german-tri, 2016-12-19:
> The JDK 8 version will break Jenkins.

@jamiesnape Could you elaborate on why this is so?

  > I installed Bazel using JDK8 on Trusty and this is making the builds not happy (e.g. https://drake-jenkins.csail.mit.edu/view/Bazel/job/linux-gcc-bazel-experimental/56/console). 

@BetsyMcPhail In that log, Bazel (?) complains `Java version is 1.7.0_111 while at least 1.8 is needed.`, which doesn't square with your comment.  Did you mean to post a different link?

jamiesnape, 2016-12-19:
> @jamiesnape Could you elaborate on why this is so?

We cannot have two different JVM versions. We can switch everything to JDK 8, but mixed will not work.

jamiesnape, 2016-12-19:
Though Jenkins recommends OpenJDK over Oracle, and there is no OpenJDK 8 for 14.04.

david-german-tri, 2016-12-19:
> We cannot have two different JVM versions. We can switch everything to JDK 8, but mixed will not work

Got it.  Do you know of any reason we should hesitate to bump to JDK 8?

david-german-tri, 2016-12-19:
> Though Jenkins recommends OpenJDK over Oracle, and there is no OpenJDK 8 for 14.04.

Oh, are we talking about Jenkins itself?  I didn't realize Jenkins and Drake tests actually ever ran on the same machine.

jamiesnape, 2016-12-19:
The Jenkins slave.

jamiesnape, 2016-12-19:
Other than the above, I guess there will be a mismatch with the MATLAB Java version, but we are already living with that for OS X.

So really, we should be fine bumping the version. It is not perfect, but I do not see huge issues.

david-german-tri, 2016-12-21:
OK, let's go ahead and bump to JDK 8 then.  It would be nice to get that done today if possible, so that we can unblock some Bazel PRs.  @BetsyMcPhail do you have some time to work on it?

BetsyMcPhail, 2016-12-21:
I'll work on it this afternoon.

BetsyMcPhail, 2016-12-21:
Trusty AMI has been update to JDK 8 and Bazel 0.4.2. Jenkins has been updated to use the new AMI.

Experimental build looks good: https://drake-jenkins.csail.mit.edu/view/Bazel/job/linux-gcc-bazel-experimental/84/

david-german-tri, 2016-12-21:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-bazel-experimental/85/console still says 0.4.0-jdk7?

jamiesnape, 2016-12-21:
It will take a while for all the instances to switch over.

david-german-tri, 2017-01-05:
This is done!  :confetti_ball: Thanks @BetsyMcPhail and @jamiesnape

closed 2017-01-05 by david-german-tri