RobotLocomotion/drake #4534: Investigate bazel support for Xcode

RussTedrake, 2016-12-17:
posted initially in #4485, but it deserves it's own issue.

fwiw, just lost a few hours trying Tulsi with bazel to work with Xcode. Long story short, I think it's not going to support our workflow without some work. I made some local edits to the tulsi code to get fairly far along (https://github.com/RussTedrake/tulsi), but it is not able to understand the "@glib://" deps, which apparently comes in through pkg-config (which comes in through the thirdparty bazel support).

The local edits that I did make were needed for it to even consider cc_* as valid targets. It was only looking for e.g. ios_application. So it's clearly not being used for our workflow yet.

RussTedrake, 2016-12-18:
looking a little more, I think that the pkg config scripts in tools/kythe, etc write temporary BUILD files, which disappear (shortly?) after bazel runs.  my best guess is that tulsi is trying to parse those files after they fail to exist.  Played a bit with making a local version of the glib-2.0 BUILD file (that could work), but have to give up for now.

david-german-tri, 2017-04-03:
I believe this has been superseded by the CLion support from https://github.com/bazelbuild/intellij, as documented at http://drake.mit.edu/clion.html

closed 2017-04-03 by david-german-tri