RobotLocomotion/drake #4535: Issue copying state of type State<AutoDiffX>

edrumwri, 2016-12-18:
I am attempting to do the following:

If I add the following function to ContinuousState:

```
  void SetFrom(const ContinuousState<AutoDiffXd>& other) {
    DRAKE_DEMAND(size() == other.size());
    DRAKE_DEMAND(get_generalized_position().size() ==
        other.get_generalized_position().size());
    DRAKE_DEMAND(get_generalized_velocity().size() ==
        other.get_generalized_velocity().size());
    DRAKE_DEMAND(get_misc_continuous_state().size() ==
        other.get_misc_continuous_state().size());
    state_->SetFromVector(other.CopyToVector());
  }

```
and the following function to State:
```
  void SetFrom(const State<AutoDiffXd>& other) {
    continuous_state_->SetFrom(*other.get_continuous_state());
    discrete_state_->SetFrom(*other.get_discrete_state());
    abstract_state_->CopyFrom(*other.get_abstract_state());
  }

```

I get the following syntax error when attempting to compile a Simulator<AutoDiffXd>:

```
In file included from /home/drum/drake/drake/systems/framework/continuous_state.cc:1:0:
/home/drum/drake/drake/../drake/systems/framework/continuous_state.h: In instantiation of ‘void drake::systems::ContinuousState<T>::SetFrom(const drake::systems::ContinuousState<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1> > >&) [with T = double]’:
/home/drum/drake/drake/systems/framework/continuous_state.cc:6:16:   required from here
/home/drum/drake/drake/../drake/systems/framework/continuous_state.h:168:5: error: no matching function for call to ‘drake::systems::VectorBase<double>::SetFromVector(drake::VectorX<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1> > >)’
     state_->SetFromVector(other.CopyToVector()); 
```

This issue will cause the Simulator<AutoDiffXd> instantiation to be disabled on the unrestricted updates PR (#4382).

RussTedrake, 2016-12-18:
looks like you are trying to set a VectorBase<double> from a VectorBase<AutoDiffXd> (which won't work).  But i don't think there is enough in your post for me to diagnose why you're getting there...?

jwnimmer-tri, 2016-12-18:
The (abbreviated) error says that it wants a method `VectorBase<double>::SetFromVector(VectorX<AutoDiffScalar>)` but that method doesn't exist.  Someone needs to explicitly slice the doubles out of the AutoDiff vector into a double vector.  See for example https://github.com/RobotLocomotion/drake/blob/53b88aa3fee4c9a14dfd3091416217811b8952b4/drake/systems/framework/continuous_state.h#L156 for going the other way (zipping a `double` vector into an `AutoDiff` vector).

jwnimmer-tri, 2016-12-18:
I find it odd that `State<T>::SetFrom(const State<AutoDiffXd>&)` would be required in that PR, though I haven't looked closely.  It is possible that `State<T>::Set(const State<T>&)` is actually what's desired?

TTBOMK `SetFrom` (with hard-coded `double` type) is a transmogrification-specific hook, not general-purposed code for use by e.g. scalartype-generic integrators.

jwnimmer-tri, 2016-12-18:
... and another idea that just occurred to me.  Do we really want `Set`, or would we rather `swap`, either the `unique_ptr<State>`, or the `State`-by-value?

edrumwri, 2016-12-18:
Frankly, I felt like I was fighting with the API to do something relatively
simple: I wanted to make a copy of the full state (continuous, discrete,
abstract), allow the user to modify that copy, and then copy the contents
back to the original.

Swapping pointers or values would be ok in my case, but I could feel the
API funneling me down a particular use-path (by eliminating copy
constructors, assignments, and such) but without guidance on how to
accomplish this task that is common with such data structures.

david-german-tri, 2016-12-19:
@jwnimmer-tri is right on all counts.  The code in the OP is a misuse/mis-generalization of SetFrom, which is my fault for not providing a comment.  I'll PR a fix right now, as well as some other convenience functions for assignment.

PR #4543 (2016-12-19): Add CopyFrom to State and ContinuousState.
Fixes #4535.

@jwnimmer-tri for both levels of review, in the hopes we can land this before Palo Alto wakes up.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4543)
commit ab51af38 (2016-12-19): Add CopyFrom to State and ContinuousState.

Fixes #4535.
closed 2016-12-19 by jwnimmer-tri