RobotLocomotion/drake #4542: Maliput Should Be Updated To Use Eigen Where Possible

liangfok, 2016-12-19:
# Problem Definition

In #4227 and #4389, custom logic was used to implement basic mathematical concepts like:

* `class Rot3` in `maliput/monolane/lane.h`
* `Rotate()` and `Distance()` in `maliput/api/road_geometry.c`
* numerous structs in `maliput/api/lane_data.h`

For increased robustness and to reduce lines-of-code, these custom implementations should be updated to use Eigen, the matrix library used by the rest of Drake.

jwnimmer-tri, 2017-03-11:
I would be curious to see where the _APIs_ should use Eigen but aren't vs the _implementation_.  The former is important to do earlier; the latter can come in due time.  Do we have a write-up or survey of API primitives that duplicate Eigen?

liangfok, 2019-06-08:
Migrated to downstream repo.

closed 2019-06-08 by liangfok