RobotLocomotion/drake #4551: Document CMake build "comprehensive nuke"

jwnimmer-tri, 2016-12-20:
Sometimes, a CMake-generated build gets confused or produces wrong results.  There's various lore floating around in slack land about how to fix it, especially with the case of `git submodules` being used.

Perhaps we should curate and document one good set of instructions for this, and post it into our sphinx docs.  Then we can always point devs at that best advice, instead of some good-but-not-quite variant from last week's slack or etc.

(This issue was broken out of #1973.)

jwnimmer-tri, 2016-12-20:
@sammy-tri would you like to take a crack at this (or assigning it)?  The manipulation subteam is perhaps the primary user of the CMake build, and does the most playing with externals and such that breaks this?

jwnimmer-tri, 2016-12-20:
Slightly related #4454 and #4504.

liangfok, 2016-12-20:
This is a great idea. At @naveenoid's suggestion, we placed the "comprehensive nuke" recipe for Drake + ROS here: http://drake.mit.edu/from_source_ros.html#how-to-perform-a-clean-build. It would be great to document the Drake-only recipe.

jwnimmer-tri, 2016-12-20:
Ah yes.  I thought I remembered being discussed, but failed to find that write-up.

jamiesnape, 2016-12-21:
Level 1 nuke for non-MATLAB and non-Python users:
```
rm -rf build
```

Level 2 nuke:
```
rm -rf build
git clean -xfd
git submodule foreach --recursive git clean -xfd
```

Level 3 nuke:
```
rm -rf build
git clean -xfd
git submodule foreach --recursive git clean -xfd
git reset --hard
git submodule foreach --recursive git reset --hard
```

PR #7134 (2017-09-29): Remove submodules and consolidate CMake options
Closes #3499, closes #3628, closes #4034, closes #4451, closes #4551.

Ultimately, the CMake externals will be removed, but it will take a little while to get the MATLAB bindings coexisting fully with the Bazel, and this removes the # 1 cause of CI flakes and saves some CI resources.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7134)
closed 2017-09-29 by jamiesnape