RobotLocomotion/drake #4552: Rename robotInterfaces, QP, ZMP stuff to follow cppguide

jwnimmer-tri, 2016-12-20:
Broken out from #1965: we should rename the remaining C++ files to match the C++ coding standard.

Checklist, per `drake-distro$ find drake ros -name thirdParty -prune -o -name examples -prune -o -name '*system1' -prune -o -name matlab -prune -o -name util -prune -o -name '*[A-Z]*.h' -print -o -name '*[A-Z]*.cc' -print -o -name '*.cpp' -print`:
- [ ] drake/systems/robotInterfaces/QPLocomotionPlan.h
- [ ] drake/systems/robotInterfaces/Side.cpp
- [ ] drake/systems/robotInterfaces/Side.h
- [ ] drake/systems/robotInterfaces/QPLocomotionPlan.cpp
- [ ] drake/systems/robotInterfaces/BodyMotionData.h
- [ ] drake/systems/robotInterfaces/BodyMotionData.cpp
- [ ] drake/systems/controllers/controlUtil.h
- [ ] drake/systems/controllers/controlUtil.cpp
- [ ] drake/systems/controllers/zmpUtil.h
- [ ] drake/systems/controllers/zmpCoMObserverStateVisualizer.cpp
- [ ] drake/systems/controllers/zmpUtil.cpp
- [ ] drake/systems/controllers/InstantaneousQPController.h
- [ ] drake/systems/controllers/InstantaneousQPController.cpp
- [ ] drake/systems/controllers/QPCommon.h

jwnimmer-tri, 2016-12-20:
@siyuanfeng-tri I think these are heavily related to your current work.  Any advice on what we should summarily rename now (or delete now?), versus waiting for some near-term walking PRs to improve on or replace the files or functionality?

I know at least most of `robotInterfaces` for example is only used by Matlab code (with the exception being the famous `Side.h` that is widely used in our C++ libraries).

siyuanfeng-tri, 2016-12-20:
I have deliberately avoided having any dependency on these files with the exception of Side. And my intention was to eliminate them all together at the end. The humanoids stack have a strong dependency on these, but I think most tests are disabled at the moment anyway. The manip IK stack might have some dependency on these, @sammy-tri / @hongkai-dai probably know better. 
I could try to just remove them and see if that breaks anything over Christmas, when I think CI will be lightly used.

jwnimmer-tri, 2017-06-19:
Re-ping.