RobotLocomotion/drake #4555: Grab bag of collision geometry artifacts - spheres and boxes

SeanCurtis-TRI, 2016-12-20:
There are some undocumented "quirks" in the current state of the drake code.  Assuming that they exist for  a reason and that the answers are out there in people's heads, I'm posing some questions so that the answers can be captured and put in the code.

1. When we define a *box* collision element, we change it from a `btBoxShape` to an equivalent `btConvexHull`.  Why?
2. All zero-radius collision spheres defined in both urdf and sdf files get automatically inflated to a minimum radius (which happens to be 1e-7).  Why?
3. The "no-margin" version of collision elements for the Bullet engine have a non-zero margin of 1e-9.  Why?  Discussions in Bullet suggest that this might be too small (see [here](https://github.com/bulletphysics/bullet3/issues/888)).

RussTedrake, 2016-12-21:
@avalenzu and @gizatt are the experts on these.

gizatt, 2016-12-24:
I'm not going to be super useful here: (1) and (2) aren't something I know anything about. When I added the no-margin version of various collision elements, I believe I set it to have nonzero-but-tiny margin to line up with the zero-radius collision spheres case (but I don't remember why I would have chosen 1E-9). I am thus totally unopposed to those things being changed and improved!

SeanCurtis-TRI, 2017-01-13:
Dynamics team status - what it will take to close this:

This will require some extensive experimentation with Bullet.  We would need to understand the impact of these design issues.  This work is unlikely to happen soon because:

1) Generally, things work.  It is possible that we're paying a performance (box to convex hull) or precision issue (sphere's with minimum radius).  But beyond those subtle issues, things work.
2) We will be bringing FCL online.  Bringing in an alternative engine and putting it through its paces is more important than polishing the bullet rock.

So, this will be deferred a long time, but should remain open.  The long range plan is to use a combination of whatever collision detection will provide the best results for us.  It is distinctly possible that Bullet will be of use to us in the future.  In that case, we want to know of these outstanding, but minor, issues.

SeanCurtis-TRI, 2019-03-25:
There is no longer any plan to use bullet. So, this is being closed. If bullet comes back into the mix, this can be re-opened.

closed 2019-03-25 by SeanCurtis-TRI