RobotLocomotion/drake #4556: Unit Tests Are Dynamically Linking Against Installed Libraries Rather Than The Newest Libraries

liangfok, 2016-12-20:
# Problem Definition

When running a unit test, we often want it to dynamically linked against the _latest_ version of the library (located in `drake-distro/build/drake/...`), not necessarily the version that was _installed_, which may be older (located in `drake-distro/build/install/...`). A common use case is we want to instrument the Device Under Test (DUT) with temporary debug print statements. By linking against the installed library rather than the most recent library, these print statements will not show up, resulting in confusion.

# How to Replicate the Problem

The following was tested on an Ubuntu 14.04 system.

Starting from a clean build environment, execute the following:

```
cd drake-distro
mkdir build; cd build
cmake .. -G Ninja -DCMAKE_BUILD_TYPE:STRING=Debug 
ninja
cd drake
ctest -VV -R package_map_test
```

You should see that the unit test `package_map_test` successfully ran.

Next, open `drake-distro/drake/multibody/parsers/package_map.cc` and modify method `Contains()` to be:

```
bool PackageMap::Contains(const string& package_name) const {
  drake::log()->info("Checking package_name {}.", package_name);
  return map_.find(package_name) != map_.end();
}
```

Note the newly added log statement, which results in a message being printed to the console.

Save and exit your text editor, then execute the following commands:

```
cd drake-distro/build/drake
ninja
ctest -VV -R package_map_test
```

You should see that the modification did not apply (nothing new is printed to the screen).

To resolve the problem, we need to build at the super-build level:

```
cd drake-distro/build
ninja
cd drake
ctest -VV -R package_map_test
```

You should not see that the newly added print statement results in a bunch of text being printed to the console.

jamiesnape, 2016-12-21:
Aside: We are almost at a stage where an install should not be required. That would be punting on this issue, of course.

david-german-tri, 2016-12-21:
@liangfok The instructions do not guarantee correctness for any build other than the superbuild, so I'm going to downgrade this to medium priority.  That said, I agree it's annoying and I don't totally understand why it's happening.

> Aside: We are almost at a stage where an install should not be required. That would be punting on this issue, of course.

I'm absolutely on board with removing the install step, and considering that a good-enough resolution.  What are the remaining obstacles?

jamiesnape, 2016-12-21:
Removing pods.cmake and SWIG, I think. The removing pods.cmake is mostly done. The SWIG part, I have not thought about yet. I doubt it would be difficult.

david-german-tri, 2016-12-21:
Right, I believe SWIG belongs to @mwoehlke-kitware on #4486.

jamiesnape, 2016-12-21:
Yes, though I think MATLAB would probably have to stay with SWIG and the logic changed not to require an install.

jamiesnape, 2016-12-21:
The only caveat, is that for now Director may need an install of Drake. That can be fixed.

david-german-tri, 2016-12-21:
Actually, I can't reproduce the original problem.  My `ninja` build is linking the `package_map_test` against `multibody/parsers/libdrakeMultibodyParsers.so`, not the one in `install/lib`.

That said, we've had all sorts of non-hermetic `install` related nightmares in the past, so this is probably another of those, and all the more reason to burn it.

liangfok, 2016-12-21:
After a F2F meeting with @david-german-tri, using `ninja -v`, we isolated the problem to be in the build command's `rpath` argument. 

Here is @david-german-tri's command:

```
[9/82] : && /usr/bin/clang++-3.9   -Werror=all -Werror=ignored-qualifiers -Werror=overloaded-virtual -Werror=shadow -Werror=inconsistent-missing-override -Werror=sign-compare -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -g  -Wl,--no-undefined multibody/parsers/test/package_map_test/CMakeFiles/package_map_test.dir/package_map_test.cc.o  -o multibody/parsers/test/package_map_test/package_map_test  /home/dgerman/drake/drake-distro/build/install/lib/libgtest.so /home/dgerman/drake/drake-distro/build/install/lib/libgtest_main.so multibody/parsers/libdrakeMultibodyParsers.so /home/dgerman/drake/drake-distro/build/install/lib/libgtest.so multibody/libdrakeRBM.so multibody/collision/libdrakeCollision.so /home/dgerman/drake/drake-distro/build/install/lib/libBulletDynamics.so /home/dgerman/drake/drake-distro/build/install/lib/libBulletCollision.so /home/dgerman/drake/drake-distro/build/install/lib/libLinearMath.so /home/dgerman/drake/drake-distro/build/install/lib/libBulletSoftBody.so multibody/shapes/libdrakeShapes.so multibody/joints/libdrakeJoints.so util/libdrakeGeometryUtil.so math/libdrakeMath.so common/libdrakeCommon.so /home/dgerman/drake/drake-distro/build/install/lib/libgflags.so.2.2.0 -lpthread thirdParty/bsd/spruce/libspruce.so thirdParty/zlib/tinyxml2/libtinyxml2.so -Wl,-rpath,/home/dgerman/drake/drake-distro/build/drake/multibody/parsers:/home/dgerman/drake/drake-distro/build/drake/multibody:/home/dgerman/drake/drake-distro/build/drake/multibody/collision:/home/dgerman/drake/drake-distro/build/drake/multibody/shapes:/home/dgerman/drake/drake-distro/build/drake/multibody/joints:/home/dgerman/drake/drake-distro/build/drake/util:/home/dgerman/drake/drake-distro/build/drake/math:/home/dgerman/drake/drake-distro/build/drake/common:/home/dgerman/drake/drake-distro/build/drake/thirdParty/bsd/spruce:/home/dgerman/drake/drake-distro/build/drake/thirdParty/zlib/tinyxml2:/home/dgerman/drake/drake-distro/build/install/lib && :
```

Here is my command:

```
[11/76] : && /usr/bin/clang++-3.9   -Werror=all -Werror=ignored-qualifiers -Werror=overloaded-virtual -Werror=shadow -Werror=inconsistent-missing-override -Werror=sign-compare -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -g  -Wl,--no-undefined multibody/parsers/test/package_map_test/CMakeFiles/package_map_test.dir/package_map_test.cc.o  -o multibody/parsers/test/package_map_test/package_map_test  /home/liang/dev/drake-distro-4/build/install/lib/libgtest.so /home/liang/dev/drake-distro-4/build/install/lib/libgtest_main.so multibody/parsers/libdrakeMultibodyParsers.so /home/liang/dev/drake-distro-4/build/install/lib/libgtest.so multibody/libdrakeRBM.so multibody/collision/libdrakeCollision.so /home/liang/dev/drake-distro-4/build/install/lib/libBulletDynamics.so /home/liang/dev/drake-distro-4/build/install/lib/libBulletCollision.so /home/liang/dev/drake-distro-4/build/install/lib/libLinearMath.so /home/liang/dev/drake-distro-4/build/install/lib/libBulletSoftBody.so multibody/shapes/libdrakeShapes.so multibody/joints/libdrakeJoints.so util/libdrakeGeometryUtil.so math/libdrakeMath.so common/libdrakeCommon.so /home/liang/dev/drake-distro-4/build/install/lib/libgflags.so.2.2.0 -lpthread thirdParty/bsd/spruce/libspruce.so thirdParty/zlib/tinyxml2/libtinyxml2.so -Wl,-rpath,/home/liang/dev/drake-distro-4/build/install/lib:/home/liang/dev/drake-distro-4/build/drake/multibody/parsers:/home/liang/dev/drake-distro-4/build/drake/multibody:/home/liang/dev/drake-distro-4/build/drake/multibody/collision:/home/liang/dev/drake-distro-4/build/drake/multibody/shapes:/home/liang/dev/drake-distro-4/build/drake/multibody/joints:/home/liang/dev/drake-distro-4/build/drake/util:/home/liang/dev/drake-distro-4/build/drake/math:/home/liang/dev/drake-distro-4/build/drake/common:/home/liang/dev/drake-distro-4/build/drake/thirdParty/bsd/spruce:/home/liang/dev/drake-distro-4/build/drake/thirdParty/zlib/tinyxml2 && :
```

Using `meld`, we can see the differences:

[image]

Basically, the problem is on my computer, `/home/liang/dev/drake-distro/build/install/lib` appears first in the `-rpath` argument list whereas on @david-german-tri's computer it appears last.

david-german-tri, 2016-12-21:
^ Which is some kind of super weird CMake behavior, I guess?

jamiesnape, 2016-12-21:
CMake is ultimately responsible, but the RPATH is apparently being "[abused](https://github.com/RobotLocomotion/drake/blob/master/cmake/pods.cmake#L31)" by PODs.

jwnimmer-tri, 2017-02-08:
With #2482 closed, I wonder if this issue has an updated disposition?

liangfok, 2017-02-08:
I will retest tomorrow using my workstation in CAM2, which was used to replicate the test described above.

liangfok, 2017-02-14:
I retested this on my Ubuntu 14.04 workstation using cf497e8e29c5a4eaa96c80c2f83616a098e4e543 and was able to replicate the issue. This issue shall thus remain open. :sob:

david-german-tri, 2017-04-13:
@jamiesnape, I've reassigned this to you, and downgraded the priority to "low" since it's CMake-specific.

jamiesnape, 2017-11-15:
@liangfok Now that we use Bazel to build Drake, is this still an issue?

EricCousineau-TRI, 2018-02-08:
Closing, as this no longer seems relevant per Jamie's comment.

closed 2018-02-08 by EricCousineau-TRI