RobotLocomotion/drake #4559: Refactor MathematicalProgramSolverInterface

hongkai-dai, 2016-12-20:
<h1> Current implementation </h1>

Currently we have two classes: `MathematicalProgram` and `MathematicalProgramSolverInterface`. Their structure are as follows

1. A flat class `MathematicalProgram` that stores the constraints, costs and decision variables.
   [image]

2. A base class `MathematicalProgramSolverInterface` and its derived class for each solver. In the `Solve` function of each derived class, it parses the constraints and costs stored in `MathematicalProgram` to the format specific to each solver, and then call the `optimize` function of each solver (`MSK_optimizetrm` for Mosek, `snopta_` for Snopt).
[image]

<h1> Challenge </h1>

I think the problem now is that inside the `Solve` function of each solver, like `MosekSolver::Solve(MathematicalProgram& prog)`, it is doing two different things

1. Parse the constraints/costs to Mosek format.
2. Call the `optimize` function of the solver (`MSK_optimizetrm` in this case).

Basically each derived solver class is just an interface, that does not have its internal data. So every time given a `MathematicalProgram prog`, the `Solve` function parses the program to the solver format, and then discards the parsed problem after `Solve` finishes.

This causes the following problems:

1. If we change the solver option (such as optimal tolerance), but with the same optimization program, then `Solve` function still needs to parse every constraint and cost, allocate memory for these constraints/costs, although the problem does not change.
2. In the model predictive control, we might change a few constraints, while leaving the rest of the optimization program unchanged. In the current settings, we still need to parse all these unchanged constraint.
3. After calling the `optimize` function of each solver, we only return an `SolutionResult` enum, which only says if the program converges or not. A lot of the solver-specific information are discarded, such as the solver's response code (info 41 in SNOPT), lagrangian multipliers, active constraints, etc.

<h1> Proposed solution </h1>

I propose to store the solver-specific data inside each derived solver class (`MosekSolver, SnoptSolver, etc`). So the `MathematicalProgramSolverInterface` would look like
```C++
class MathematicalProgramSolverInterface {
 public:
  MathematicalProgramSolverInterface() : prog_(nullptr) {}
  MathematicalProgramSolverInterface(const MathematicalProgram& prog) : prog_(&prog) {}
  virtual SolutionResult Solve() = 0; // Solve the optimization program stored in prog_
  virtual SolutionResult Solve(MathematicalProgram* prog) = 0; // Solve the optimization program given an MathematicalProgram object.
 private:
  MathematicalProgram* prog_; // This pointer does NOT own the MathematicalProgram object. 
}
```

And the derived solver class would look like
```C++
class MosekSolver : public MathematicalProgramSolverInterface {
 public:
  MosekSolver() : MathematicalProgramSolverInterface(), task_(nullptr), env_(nullptr), rescode(MSK_RES_OK) {}
  MosekSolver(const MathematicalProgram& prog) : MathematicalProgramSolverInterface(prog) {
    // Parses the constraints, costs, decision variables.
  }
  SolutionResult Solve(MathematicalProgram* prog) override {
  // Parses the constraints, costs, decision variables.
  // Call Mosek optimize function.
  }

  SolutionResult Solve() override {
  // Call Mosek optimize function.
  }
  
 MSKrescodee response_code() const {return rescode_;}

 private:
  MSKtask_t task_; // Mosek task, contains the parsed decision variables, costs, constraints, etc.
  MSKenv_t env_; // Mosek environment, contains the solver certificate, etc.
  MSKrescodee rescode_; // Mosek response code, useful for debugging.
}
```
Namely we still retain the original interface, that a solver class can call `Solve` function, by passing in a `MathematicalProgram` object. But we can also construct a solver class object from `MathematicalProgram`, and then call the `Solve` function for the stored `prog_`. Notice that we can also access the solver-specific data from `response_code()` function. We can add more accessor functions for each specific solver.

hongkai-dai, 2016-12-20:
@RussTedrake and @sammy-tri what do you think of this?

hongkai-dai, 2016-12-21:
Unfortunately there is a bad problem coming out from the design above, shown in the following example
```C++
MathematicalProgram prog;
prog.AddDecisionVariable(...);
prog.AddConstraint(...);
prog.AddCost(...);
// Now construct the solver from the current prog
MosekSolver mosek_solver(prog);
// Add another constraint
prog.AddConstraint(constraint_new);
// oops, mosek_solver has not parsed the newly added constraint constraint_new yet.
mosek_solver.Solve();
```
Namely, mosek_solver does not know if the program that has been parsed into the Mosek format, is the same as the program that `mosek_solver.prog_` points to.

siyuanfeng-tri, 2016-12-21:
prog.updateconstarints also suffers the same problem you mentioned.
you could have a dirty flag in the prog for all the constaints / costs. whenever you call solve, the solver can update / add the new stuff.

hongkai-dai, 2016-12-21:
Thanks for the comments, but that does not really solve the problem, consider the following example with two `MosekSolver` objects
```C++
MathematicalProgram prog;
prog.AddDecisionVariable(...);
prog.AddConstraint(...);
MosekSolver mosek_solver1(prog);
MosekSolver mosek_solver2(prog);
prog.AddConstraint(...); // Now the dirty flag should be true.
mosek_solver1.Solve(); // parsing the program again, and the dirty flag should be false after the program is parsed.
mosek_solver2.Solve(); // oops, the dirty flag is false, the newly added constraint is not parsed into mosek_solver2.
```

siyuanfeng-tri, 2016-12-21:
hmm, this is getting more and more complicated. so the prog needs separate dirty flags for each solver... and that probably requires the solver constructor to take non const ptr to the prog, so that the solver can register itself in prog's list of solvers... this is getting very ugly.

jwnimmer-tri, 2016-12-27:
How much of a performance hit is the re-setup of the constraints for every new solve?

This is just an "incremental recompilation" style of problem.  I can think of a few designs that help with either memoizing repeated setup computations, or handling a "recompile only what changed" ripple.  (Without the slicing / manual dirty bit errors shown in the upthread designs.)  But they will all make the code more complicated, more costly to develop and maintain, etc.

If the answer is just "buy a better computer" or "actually who cares" then don't need to solve this issue in code.  If that's still not good enough, the next step is to commit a performance benchmark & regression (ideally via CI), establish a baseline (and profile for what's the hotspot), and _then_ optimize the code to perform better subject to the measurement regime that we've decided matters.

hongkai-dai, 2016-12-27:
@jwnimmer-tri , I have not run the profile to check the time spent on each constraint. But we want to remove any redundant computation in optimization, since when controlling the robot, the optimization should run as fast as possible (~1000Hz).

The purpose of this cleanup is not only to speed up parsing the constraints, but also to store solver result inside each `MathematicalProgramSolverInterface` object, so that the user can diagnose the optimization problem. Currently we offer very limited diagnostic information, since `MathematicalProgramSolverInterface` does not store any data.

jwnimmer-tri, 2016-12-27:
Yes, I fully support offering extra data after a Solve.  It seems like we could solve that relatively separately from the re-parsing optimizations.

For reference, the rigid body plant has a `contact_detail` concept that solves a similar problem of offering supplemental data after a computation, depending on which backend was used.

hongkai-dai, 2019-01-25:
Close this issue as we have the new issue #9633 describing refactoring MathematicalProgram.

closed 2019-01-25 by hongkai-dai