RobotLocomotion/drake #4569: Doxygen Website Incorrectly Renders Inheritance Diagram for Maliput / Monolane Classes

liangfok, 2016-12-21:
# Problem Definition

Doxygen generates inheritance diagrams for classes in Drake's documentation website. The inheritance diagrams for all of `drake-distro/drake/automotive/maliput/monolane/` is broken. See, for example, the inheritance diagram for `drake::maliput::monolane::RoadGeometry`:

[image]

It should be pointing to `drake::maliput::api::RoadGeometry` but is instead pointing to itself.

For more background information, see: https://reviewable.io/reviews/robotlocomotion/drake/4389#-KZSZs5XVm1fAZIkSZI_

jwnimmer-tri, 2016-12-23:
http://drake.mit.edu/doxygen_cxx/classdrake_1_1maliput_1_1monolane_1_1_road_geometry.html

In the code that went into master, the class overview now has `api::RoadGeometry` hyperlinked, very nearby to the figure.  I think that's good enough.

closed 2016-12-23 by jwnimmer-tri

soonho-tri, 2017-03-02:
It seems `HIDE_SCOPE_NAMES` is a related doxygen-config option:

> If the HIDE_SCOPE_NAMES tag is set to NO then doxygen will show members with their full class and namespace scopes in the documentation. If set to YES, the scope will be hidden.

From http://www.stack.nl/~dimitri/doxygen/manual/config.html