RobotLocomotion/drake #4574: Consider removing drake/automotive CMake integration

jwnimmer-tri, 2016-12-21:
Can anyone think of a good reason to keep the `CMakeLists.txt` support for `drake/automotive`?

I think I'm one PR from having all of `drake/automotive` tested and demo-able under the Bazel instructions.  In the future, once #4131 and #4303 are done we would end up shipping the library and message portions of `drake/automotive` as part of an overall `libDrake`.  Still, many demos would be documented and buildable only via Bazel.

So once we have Bazel ready, and confirmed working for the automotive development team, is there any reason I can't just delete the `CMakeLists.txt` support (and thus simplify our dev cycle to worry about one less build system most of the time)?

liangfok, 2016-12-21:
I suppose the question of "what bugs will Bazel let through that CMake will not let through" is relevant to this discussion.

Another question is: how easily will it be to add `CMakeLists.txt` back to `drake/automotive` should we, at some point in the future, encounter a customer that demands it?

jamiesnape, 2016-12-22:
> So once we have Bazel ready, and confirmed working for the automotive development team, is there any reason I can't just delete the CMakeLists.txt support (and thus simplify our dev cycle to worry about one less build system most of the time)?

Technically, this an open-source project, so there may be other users now or in future (whether customers or otherwise).

jwnimmer-tri, 2016-12-24:
> I suppose the question of "what bugs will Bazel let through that CMake will not let through" is relevant to this discussion.

I don't think we know, but I haven't been able to think of any meaningful examples.

The biggest related risk is that when #4303 integration happens, we will be publishing our `automotive` library into CMake land, but the `automotive` developers will not be using CMake regularly.  Thus, bugs in the Bazel -> CMake library integration may go undetected for longer.  I hope we can mitigate that by (1) as #4303 moves along, ensuring that there are enough regression tests in-tree to ensure correctness, and (2) having `libdrake.so` pull in many things at once (`automotive` along with `systems`, `solvers`, etc.), so that if any one thing breaks it will probably all break and somebody will notice (and then fix up (1)'s tests to prevent that regression again).

> Another question is: how easily will it be to add CMakeLists.txt back to drake/automotive should we, at some point in the future, encounter a customer that demands it?

Since we will publish our library code via CMake rules that shell out to Bazel, the case for when this would be needed is unclear.  We could even publish our compiled binaries (`bazel-bin/**`) as part of a nightly / monthly Drake release via Bazel and/or CMake if we wish.

Thus, the only use case I can think of is developers who want to contribute new code to `automotive`, but want to only ever invoke CMake (which shells out to Bazel, and is thus maybe annoying to use during active development) instead of using Bazel directly.  I don't think I would add back `CMakeLists.txt` for just that case.

> Technically, this an open-source project, so there may be other users now or in future (whether customers or otherwise).

My proposal is to say "Apologies, but we're Under Construction" to any current users of `automotive` via CMake, and then to re-support them with the #4303 integration + `libdrake.so` rollup library.  (We can't do this out of the gate, because TTBOMK the #4303 has to happen in bottom-up dependency order, so we have to do `common` etc. first, then `systems` etc., then `automotive`.)

Part of the purpose of this issue is to see if anyone in open-source land would like to vote for keeping CMake support around in the interim, until we finish the build-system-glue tasks.

So far, it sounds like "Drop CMake support in `automotive` in order to move faster" has no strong objections, but I'm still listening in case there are.

jamiesnape, 2016-12-24:
> My proposal is to say "Apologies, but we're Under Construction" to any current users of automotive via CMake, and then to re-support them with the #4303 integration + libdrake.so rollup library. (We [...]

That is not the usual practice for building an open-source community, and especially in the early days people will not trust this project and move on. A vibrant open-source project is to an extent always "under construction", so it is not a good argument. There are always accommodations that have to be made when deciding that a project should be open-source. We already have people forking the project instead of using it and contributing back. Equally, we are now forking the build systems of other projects (the whole `tools` folder) and not contributing back to them.

> So far, it sounds like "Drop CMake support in automotive in order to move faster" has no strong objections, but I'm still listening in case there are.

Consider my objections extremely strong for what it is worth. I am more bothered about "dropping XXX support because YYY does not use it" than the specifics of what or who XXX and YYYY are.

liangfok, 2016-12-26:
If we remove `CMakeLists.txt` from `drake/automotive`, will the libraries and header files within `drake/automotive` still be accessible to other ROS packages via Catkin? I ask because there are potential users who are patiently waiting for Drake automotive simulations with ROS bindings.

jamiesnape, 2016-12-26:
No, they will not. We build Drake for ROS with Catkin itself, which requires CMake.

jwnimmer-tri, 2016-12-26:
Jamie's message a couple days ago made me revisit what I'm trying to accomplish here, and thus revise my proposal.  My goal is to remove friction from development velocity.  In that light, I think that keeping the vanilla CMake rules around (declare a library; declare some C++ unit tests) is not a big deal.  Updating the list of source files only takes a moment.  The other occasional pain is adding / conditional-izing the externals, which is infrequent.  Keeping the CMake libraries around will also help CLion users continue working with `automotive` until we have #4485 under control, and help any downstream that is using or wants to use CMake supported across the #4303 transition without any gaps.

For my new proposal, I would like to drop support for running the more complicated tests and experiments via CMake land (e.g., anything involving python, or the bot-foo tools, etc.).  Those are what waste hours and days trying to get and keep working, so dropping them gets us 99% of the benefit.

Specifically, my proposal is that `automotive/README.md` only explains how to use the demos using the Bazel build system, and that any automated test that is difficult to get working in both Bazel and CMake is allowed to be PRd as Bazel-only.

jwnimmer-tri, 2016-12-27:
... and a quick reply to a side discussion as well:

> [W]e are now forking the build systems of other projects (the whole `tools` folder) and not contributing back to them. Projects may or may not want Bazel, but there should at least be a roadmap to try to upstream.

First, let's be clear that we're already open-sourcing our nascent `otherlibrary.BUILD` files, but we're hosting them in our tree.  Someone who wants to use Bazel with one of our externals has a license grant to use the `BUILD` files we've written.  Other projects that use Bazel as a tool aren't even doing that much for their open-source externals.  We should be thankful for this much, at least.

But of course you're right that by doing it in Drake's tree instead of the externals' trees, the Bazel-related files are less immediately useful for / usable by other parties, more difficult for us to maintain over time, etc. etc.

I agree that we should be pushing Bazel support to upstreams.  My mindset so far has been to get it working extremely well in Drake with our `otherproject.BUILD` files, and then once we've learned the best Bazel choices and are happy living in that world, we can work with upstreams to contribute our work back.  Right now, it seems like its changing fast and we don't quite have all of the best practices down pat yet.

It's also not super useful to contribute back only the `BUILD` files.  We should also help upstreams get their own `WORKSPACE` file set up as well, and have their Bazel builds covered by CI.  (This could also help us, once Bazel supports chained WORKSPACE files.)  For now, I think using our CI with externals' specific sha snapshots of source code is most effective.

In short, I would _absolute love it_ for upstreams to have native Bazel support.  I think TRI helping to lead the charge there by PR'ing it would be a meaningful and lasting contribution to open-source robotics.  I think it's just more effective to get there by baking all the files in our tree first, and _then_ pushing them to upstreams with a nice bow on top.

jamiesnape, 2016-12-27:
> Keeping the CMake libraries around will also help CLion users continue working with automotive until we have #4485 under control, and help any downstream that is using or wants to use CMake supported across the #4303 transition without any gaps.

I am kind of thinking we should skip #4303, get everything ready and then make one switch, but that is discussion for elsewhere.

> For my new proposal, I would like to drop support for running the more complicated tests and experiments via CMake land (e.g., anything involving python, or the bot-foo tools, etc.). Those are what waste hours and days trying to get and keep working, so dropping them gets us 99% of the benefit.

Sounds good.

jwnimmer-tri, 2016-12-28:
So far, it sounds like consensus.

Next steps: I have a feature branch with `automotive_demo.py` in Bazel, I'll work on getting that PR'd in chunks.  We can live with it for a couple weeks and if we're still happy, commit a `README.md` update that resolves this issue.

PR #4737 (2017-01-12): Port automotive_demo.py launcher to Bazel
This PR is (1) build system changes, (2) source code changes at its touch-point with the build system, and (3) documentation updates.

I have tested it extensively locally, but reviewers should definitely pull this code and test for themselves as well.

Closes #4574.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4737)
closed 2017-01-12 by jwnimmer-tri