RobotLocomotion/drake #4578: system_identification_test is slow

jwnimmer-tri, 2016-12-21:
The `drake/solvers/system_identification_test` is slow -- approx 40 seconds on a release build on modern hardware.  When running in debug mode or under valgrind, it takes minutes.

My understanding is that the slowness is due to perhaps only some of the cases, as if one of them is some kind of acceptance test, instead of a unit test.  I also recall hearing that the test code uses brute-force implementations of tools like the Simulator that didn't exist at the time, but do now.

Can we spend a half-day making this better somehow?  It's tied with RBT as one of the worst pain points in my local builds.

/CC @ggould-tri

ggould-tri, 2016-12-21:
The sysid test is indeed slow, because it implements a practical sysid problem (quadratic cost function over two hundred or so linear equations).  I believe that profiling will show that the time is spent in the optimizer.

The hand-coded spring-mass system simulation is a lot of lines of code that could probably be replaced, and probably should be, but don't expect this to cause a speedup (I doubt it's where the time is spent) or a reduction in lines of code (because System code is boilerplatey enough that it will doubtless be more lines of code than my special-purpose version).  It might however give a smaller epsilon due to the better differential equation solver.

To speed it up I recommend separating out the slow test into a slower test suite.  Practical system ID problems are not fast, and I don't believe that we should regard sysid as fully tested without one.  If on the other hand we want to get the result more quickly, I suspect you will want to attack that at the optimizer level, for instance by reformulating the problem as a QP.

jwnimmer-tri, 2016-12-21:
Alright, it sounds like I didn't entirely have the right picture.  Having the acceptance test (tests?) in a different file (files?) than the unit tests, marked `size="medium"`, sounds like a fine solution.

EricCousineau-TRI, 2017-04-18:
Failure on linux-xenial-gcc-bazel-continuous-debug (job [#586](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-xenial-gcc-bazel-continuous-debug/lastFailedBuild/)) due to timeout in `drake/solvers/system_identification_test`, triggered by #5858.
This happened once before on April 3.

For now, will assume these are sporadic occurrences.

jwnimmer-tri, 2018-09-21:
(Still a minor annoyance, but I agree we don't need an open issue for it.  Next time someone is tired of waiting, then can implement the fix.)