RobotLocomotion/drake #4592: RigidBodyTree::collisionRaycast() Needs To Be Made const.

liangfok, 2016-12-22:
In System 2.0, we need to call `RigidBodyTree::collisionRaycast()` from a `const` context like from within the bodies of `System::DoCalcUpdate()` and `System::DoPublish()`. Currently, we cannot do this without resorting to using `const_cast`.

sherm1, 2016-12-22:
While you're in the neighborhood, @SeanCurtis-TRI ...

SeanCurtis-TRI, 2016-12-28:
This is part of a larger problem; that being the process of querying the tree updates the tree.  We have the same issues when doing contact computation.  This will probably be deferred until `GeometryWorld` takes over these responsibilities and `GeometryQuery` provides an appropriate `const` interface.

liangfok, 2016-12-28:
Thanks for investigating. I will close this issue and trust that it will be eventually resolved once GeometryWorld and GeometryQuery are introduced.

closed 2016-12-28 by liangfok