RobotLocomotion/drake #4595: boxcar.sdf Is Missing Visual Elements

liangfok, 2016-12-23:
I didn't try to load it but from a quick glance, it appears to be missing any visual element:

https://github.com/RobotLocomotion/drake/blob/09d818b81f79cd054262605ddbe2c6959bae28a8/drake/automotive/models/boxcar.sdf

Is this true? Is this expected?

jwnimmer-tri, 2016-12-23:
@liangfok You added this model to the repo, for use by unit tests (only).  Probably you are the best person to answer whether its content is as-expected or not.

liangfok, 2016-12-23:
Oh yeah, you're right. Looks like the unit test (`automotive_simulator_test.cc`) doesn't check the model instance's visualization. It only checks the existence of a link in the model instance. I'll update this issue to reflect the need to improve the unit test.

jwnimmer-tri, 2016-12-23:
We could also just delete the boxcar(s)?

liangfok, 2016-12-23:
SGTM. Either way is OK. In the past, having BoxCars was helpful for detecting excessive overhead of using models with multiple articulated bodies and mesh visualizations.

jwnimmer-tri, 2017-01-13:
@liangfok This is a quick fix either way, yes?  Can you choose one and PR it?  Issues with 2-minute fixes shouldn't linger.

liangfok, 2017-01-13:
Yes, it's a quick fix. I'm on it.

PR #4786 (2017-01-14): Removes boxcar.sdf and the unit tests that rely on it.
Resolves #4595. I decided to remove boxcar.sdf since I believe the unit tests using the Prius SDF models are even more rigorous and the performance issues uncovered by comparing the box car model to the Prius models is now known. In addition, there is still a `boxcar.urdf` in case we do want to measure performance using a Box Car model.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4786)
closed 2017-01-14 by jwnimmer-tri